mavlink/qgroundcontrol #4502: Mission editor: (dangerous) WP parameters change is picky about editing.

AndKe, 2017-02-01:
Reproduced on today's master, (Linux), observed earlier:

How to reproduce: 
(while connected to a ArduCopter, but if not (no hold-time visible) results are still visible)

-Make a mission, consisting of at least WP1, WP2 (anywhere).
-WP2: by selecting all text, and overwriting (!): set hold time to 57s  , altitude 57m
-select WP1 , and back to WP2.
-observe that WP2 has the expected 57s, 57m.
-in WP2 altitude, place cursor between 5 and 7 , press 6 
-in WP2 hold time, place cursor between 5 and 7 , press 6 
-select WP1 , and back to WP2.
-observe that WP2 altitude is still 55m.  

When user edits text, by inserting or deleting digits, selecting and overwriting less than whole contents of the textbox, - the change is only visual (dangerous)

This behavior seems to be in most(all?)  entry boxes that have floating point numbers:
Altitude, Hover/cruise-speed,   1, In survey: (at least) , Grid Angle, Grid spacing, and Turnaround distance.

DonLakeFlyer, 2017-02-01:
This looks to be a focus problem when switching between WP1 to WP2 it doesn't trigger onEditingFinished. Need to force focus loss somehow or force onEditingFinished when you switch from one waypoint to another.

@NaterGator Are you familiar with the Qt focus handling code? Ever time I try to use it I can't get anything to work.

AndKe, 2017-02-02:
For what it's worth: it's not reproducible on Android , so it might be a Qt issue, (and be fixed in current version ?)

DonLakeFlyer, 2017-02-06:
Yeah, this is a really important fix (3.2 won't go out without it). I had heard some reports of it before but nobody had really nailed it down to a reproducible scenario. And like you say in reality it happens all over the place. So big thanks for that.

DonLakeFlyer, 2017-02-14:
No repro on OSX.  Cranking up Linux.

DonLakeFlyer, 2017-02-14:
@AndKe I can't repro on Linux either. I have a feeling I'm not reproing exactly the same. Could you make a video?

AndKe, 2017-02-14:
@DonLakeFlyer  Please see video at https://youtu.be/-KQ992RHkdo
I did load an standalone app that visualizes the keystrokes so you can see the different, futile, attempts to edit.
The app does not influence the QGC's bahavior.

in short: 
-marking all text (numbers) and overwriting works
-just placing a cursor inbetween , and inserting  an digit, or replacing a few, does not stick, even with "enter"
-same for selecting a digit (mouse or shift+arrowleft/right)  , then replacing it ...

DonLakeFlyer, 2017-02-14:
Ok, that is exactly what I was doing and it works fine. Did you build this GGC yourself or are you using a downloaded version. If downloaded, what variant? AppImage? Tarball? If built yourself are you building with QGC 5.5.1?

AndKe, 2017-02-14:
cloned from github, updated today. (you can see build on top)
it's built on the same computer as recorded on, using 
```
qmake -v
QMake version 3.0
Using Qt version 5.5.1 in /home/andre/Qt/5.5/gcc_64/lib
```
to further add to mystery, this happens with boxes containing floating point values, (or at least presented as such)
mission item "delay" (displays seconds as integer)  - behaves as expected, regardless of editing method.

DonLakeFlyer, 2017-02-14:
Could you try using an AppImage downloaded daily build to see if the same thing happens? Also are you running Ubuntu, what version?

AndKe, 2017-02-14:
It's Ubuntu 16.10 x64 , I have a virtual 16.04 I could try as well..  where do I find the image ?

..and the Qt is installed from the public http://www.qt.io/download-open-source as described in README.md. 
may it be there are some bugfixes or something ?

Finally, not entirely comparable, but APM2 , built in the same environment , at this computer , have more of a list-form of waypoints,  any method of editing sticks...  (again - I am not sure how relevant that is)
[image]

AndKe, 2017-02-14:
found appimage in releases

DonLakeFlyer, 2017-02-14:
Daily download links at bottom of page here: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html

DonLakeFlyer, 2017-02-14:
I was running on my 14.04 vm, let me try 16.10 as well.

AndKe, 2017-02-14:
Appimage 3.1.2 , is no better.

DonLakeFlyer, 2017-02-14:
Ok, I'll try to repro on 16.10

AndKe, 2017-02-14:
Appimage , on Ubuntu 16.04 LTS , same thing, I have no older at hand.

AndKe, 2017-02-14:
FYI: I installed a virtual Ubuntu 14.04 x64  - and ..appimage only , I get this:
```
andre@ubuntu:~/Downloads$ ./QGroundControl.AppImage 
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
VMware: vmw_ioctl_command error Invalid argument.
Aborted (core dumped)
andre@ubuntu:~/Downloads$ 

```

DonLakeFlyer, 2017-02-14:
You need to install gstreamer packages yourself: https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming

DonLakeFlyer, 2017-02-14:
I can't repro on 16.04 either.

AndKe, 2017-02-14:
gstreamer added, did not change the way it crashes on fresh 14.04LTS  (the crash seems to be related to VMWare, still ,very unusual...)

DonLakeFlyer, 2017-02-14:
The gstreamer packages should just remove the error. I didn't see if was crashing. I've never seen that before.

AndKe, 2017-02-14:
it's crazy that you can't see it on 16.04 - the 16.04 I have is a clean install I made - .. I defined Norwegian keyboard layout.. , regional settings.... these are floats.. will test US , or .. what do you use ?

DonLakeFlyer, 2017-02-14:
I'm running pretty clean 16.04 on Parallels using US settings. I've asked other Linux folks to try to repro.

DonLakeFlyer, 2017-02-14:
Is there some way I can duplicate you norwegian setup/settings? Tell me what to do, some file or something?

AndKe, 2017-02-14:
do you run the appimage of 3.1.2 , or have it all built there ?  the 16.04 I had have everything for compiling there, but have same result using the image .. the 14.04 is just clean + the appimage.

DonLakeFlyer, 2017-02-14:
I tried both AppImage and my own build. Both worked fine. My guess is it has something to do with your norwegian settings.

AndKe, 2017-02-14:
for Norwegian,  set "system settings" like this:

[image]
[image]

I did try to select US regional & keyboard, (in 16.04 VM) no success.

AndKe, 2017-02-14:
YES ! - selected US system setting , keyboard - it did not help, but rebooted !  - and it works - will reboot the VM now for each item, Norwegian keyboard first..

dogmaphobic, 2017-02-14:
Jumping from up high here and not really knowing what's going on. With that said, is there any float/double parsing going on? That would be something that would changed based on locale (US versus the rest of the world).

AndKe, 2017-02-14:
Nailed it :) 
Regional formats: 
[image]

.. causes the problem... 
please notice how we, (and many other countries)  use the comma vs dots differently from US..

DonLakeFlyer, 2017-02-14:
I think I know what it is. There is a Qt DoubleValidator on the field which probably doesn't understand regional formats. Take me a little while to try. In the middle of something else right now. Huge thanks for plugging away at this one for me.

DonLakeFlyer, 2017-02-14:
@AndKe What's the package to install in order to get the Norwegian format?

dogmaphobic, 2017-02-14:
If it's a locale issue, any language other than English would do it. It looks like a float parser using period instead of commas for decimal delimiters. There should be an option to define these, regardless of the current language (as in *Regional Formats* above).

NaterGator, 2017-02-14:
Is this Qt.ImhFormattedNumbersOnly + DoubleValidator + locale with commas screwing things up? I could have sworn that bug was squashed upstream a while ago.

AndKe, 2017-02-14:
@DonLakeFlyer  To get Norwegian(or any other regional settings)
In language support, there is a tab "Language" (also in screenshot above)  - click the big button "install/remove languages"   , scroll down, and check off for desired language.
Then, it becomes a choice in the right tab ("regional formats")

DonLakeFlyer, 2017-02-15:
I can repro know with Norwegian install.

DonLakeFlyer, 2017-02-15:
It's the DoubleValidator on FactTextField this is screwing up.

DonLakeFlyer, 2017-02-15:
That's not really needed since the higher ligher validation will kick in after the fact.

closed 2017-02-15 by DonLakeFlyer