mavlink/qgroundcontrol #4504: Connect generic vehicle

31Romain, 2017-02-02:
Hi !

I got a problem while I'm trying to connect a new generic vehicle on QGC.
Here is the google groups discution : https://groups.google.com/forum/#!msg/qgroundcontrol/l37Eh9FshhU/LkywkasKAwAJ

To recap the issue : I've got an autopilot who sends mavlink data to QGC (through serial com), I can grab and decode that mavlink data from QGC, but i still cant see the UAV on map (position sends by autopilot). 

The UAV still stay not connected, even if the serial com 4 is connected and used to mavlink data transfer

DonLakeFlyer, 2017-02-07:
More from Romain:

I’m not sure to understand what you mean by “support parameters” vehicle. It’s just an autopilot who send mavlink data to QGC (this part works well).

Also what about PARAM_REQUEST_LIST message ? It’s the response of vehicle to QGC initialization ? If so, I should see some transfer into the Mavlik Inspector right ? But it still stay empty.

To get all the information about the UAV (altitude, latitude, longitude …), I use qDebug() function to print it into the console after decode function, obviously.

Now I had seen something. I still use the Mavlink v.1 lib. Do you think if I update it to version 2.0, it may work ? I have already tried but I gets me a lots of code errors. So before to fix it all, I prefer to be sure that is those two things are probably related.

DonLakeFlyer, 2017-02-07:
PARAM_REQUEST_LIST is sent from QGC to your vehicle. You vehicle should respond with the set of parameters it supports. Does you vehicle not respond to the PARAM_REQUEST_LIST message when its gets it?

31Romain, 2017-02-10:
No, you found my mistake. My vehicle does not respond to the PARAM_REQUEST_LIST message for now.
So, I'm trying to decode the mavlink data send by QGC to my vehicle. 
I cannot see the PARAM_REQUEST_LIST_MESSAGE_ID (id = 21), I only get the HEARTBEAT_MESSAGE (id = 0). I just started to work on it so it's probably my fault.
Thanks for all, I will try to fix it on my own !

DonLakeFlyer, 2017-02-13:
You shouldn't need to respond to PARAM_REQUEST_LIST. There is a bug on the QGC side. But if you start to do that things will work.

closed 2017-02-15 by DonLakeFlyer