mavlink/qgroundcontrol #4506: PX4: Hobbs meter

LorenzMeier, 2017-02-02:
@dogmaphobic These two parameters implement a 64bit microseconds total flight time count (hobbs meter):

```
Flight time in microseconds = (LND_FLIGHT_T_HI << 32) | LND_FLIGHT_T_LO
```

dogmaphobic, 2017-11-10:
I assume this is milliseconds instead? The values I'm getting only make sense if it's milliseconds (5+ hours instead of 19 seconds).

LorenzMeier, 2017-11-10:
@bkueng has the details

dogmaphobic, 2017-11-10:
Nope, it's microseconds alright. The autopilot I was testing with indeed only had 19 seconds of "armed" time.

dogmaphobic, 2017-11-10:
I armed it for 10 seconds and the timer went up 10 seconds.

DonLakeFlyer, 2017-11-11:
QGC cannot currently support volatile parameters. This will causes the parameter cache hit to fail 100% of the time. Supporting volatile parameters is doable but non-trivial. Unless you want the cache to no longer work I'd suggest, either:
* Don't support this through a parameter so cache doesn't break
* Live with broken cache and create/implement new volatile parameter architecture. This won't happen before I head out of town in a week and then I"m gone for two weeks. So QGC users will have to live with full param load for a while.
* Back out firmware change until QGC supports volatile parameters so cache continues to work

DonLakeFlyer, 2017-11-11:
Also FYI: I do think adding support for volatile parameters to QGC is a good thing. It's just a matter of timing.

dogmaphobic, 2017-11-11:
Note that the parameters themselves have been there for a very long time. I've just added code to read and process them. The cache would have been broken for a very long time as well. This would explain why I felt the cache system would fail quite frequently. Any time you "arm" the vehicle, the hobbs meter increase and the parameter data changes.

NaterGator, 2017-11-11:
I noticed this a while back. I'm not sure if we have a convenient way to flag volatile params, maybe it makes sense to use a leading underscore as a volatile qualifier? We can hash everything else and explicitly request all volatile params? We could just assign them a value of 0 or NaN during hash generation so we ensure a change in any volatile parameter name is detected.

DonLakeFlyer, 2017-11-12:
> I'm not sure if we have a convenient way to flag volatile params, maybe it makes sense to use a leading underscore as a volatile qualifier?

It should be a new field for the parameter metadata on firmware side. Then add volatile to FactMetaData and then update ParameterManager to handle the flags.

> The cache would have been broken for a very long time as well.

I've been kind of wondering about that. Seems like I only see hit sproadically when I work with older firmwares from other folks.

dogmaphobic, 2017-11-12:
>I've been kind of wondering about that. Seems like I only see hit sproadically when I work with older firmwares from other folks.

Right. Same here but now there is an explanation. It will miss the cache if you "armed" the vehicle. If you don't, the cache doesn't change.

bkueng, 2017-11-13:
> It should be a new field for the parameter metadata on firmware side. Then add volatile to FactMetaData and then update ParameterManager to handle the flags.

I agree with that. We actually already have more such volatile params. There's `COM_FLIGHT_UUID` and ekf2 has some in-flight learned mag bias parameters. We generally make sure they are all changed and saved *after* disarming.

Related: https://github.com/PX4/Firmware/issues/8141

dogmaphobic, 2020-02-11:
This "issue" has been completed (quite a while ago). The issue with volatile parameters, if it is still the case is a different thing.

closed 2020-02-11 by dogmaphobic