mavlink/qgroundcontrol #4509: Flight mode menu pops up in wrong place on Android

dogmaphobic, 2017-02-03:
Current master has the flight mode menu popping up on the bottom of the screen instead of where you touched the QGCLabel with the current mode:

[image]

DonLakeFlyer, 2017-02-07:
Very strange. I thought I had screwed this up with some other toolbar stuff I did. But I don' think I touched this part. Works fine on Mac which is so odd.

DonLakeFlyer, 2017-02-26:
As far as I can tell this is a Qt android bug. All Menu qml element show up in the wrong place. They are supposed to show up right where you clicked. There is no control to position them. All you do is call Menu.popup().

DonLakeFlyer, 2017-04-13:
Qt bug: https://bugreports.qt.io/browse/QTBUG-60151

DonLakeFlyer, 2018-06-20:
Qt will fix it when they fix it

closed 2018-06-20 by DonLakeFlyer