mavlink/qgroundcontrol #4512: Cannot deploy QT on Android

kanishkanamdeo, 2017-02-04:
I have android SDK 22 installed. Ubuntu 16.04. Following is the output I get 
`Project MESSAGE: Qt version 5.5.1
Project MESSAGE: Linux build
Project MESSAGE: QGroundControl Development master:91a3494 2016-11-17 09:07:09 -0800
Project MESSAGE: Debug flavor
Project MESSAGE: Skipping support for UVC devices (conflict with Qt 5.5.1 on Ubuntu)
Project MESSAGE: Using Local QtLocation headers for Qt 5.5.1
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for XBee API (64-bit Linux builds not supported)
Project MESSAGE: Skipping support for RT-LAB (unsupported platform)
Project MESSAGE: Including support for speech output
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for video streaming
Project ERROR: Unknown module(s) in QT: serialport location-private positioning-private
20:01:17: The process "/usr/lib/x86_64-linux-gnu/qt5/bin/qmake" exited with code 3.
Error while building/deploying project qgroundcontrol (kit: Desktop)
When executing step "qmake" `

I have followed the readme well, used the online installer and all, but it has been a few hours and I'm still out of luck here!

DonLakeFlyer, 2017-02-06:
You installed Qt onto your machine using the online installer to install Qt 5.5.1 as specified in the README? If so, there are likely some options in the installer that needed to be checked that weren't. Qt folks keep changing the installer such that it install some things and not other. Use the repair functionality of the Qt installer to install additional packages named location, positioning serialport. I don't know the exact names as I type this. But I think if you look at the Qt installer you should be able to figure out what they are as they related to the error message above.

booo, 2021-10-30:
Building QGC for android works. If you still have problems please let us know.

closed 2021-10-30 by booo