mavlink/qgroundcontrol #4513: Very slow framerate on UDP video preview.

AndKe, 2017-02-05:
Let me be clear on this;:  I do not KNOW if this was caused due to the recent video-recorder commits, but I love the recording feature.  - however, I must say I suspect it, because I can't remember to have seen this issue before.  

When I start QGC (desktop), and it finds a 3DR Solo video stream (UDP), it's at a very low framerate. - I guess less than 1 FPS.
Surprisingly: the new recording feature produces very fine videos at full speed, it's only the preview-window that is slow. ( regardless of being small, or fullscreen).

Is there something I can do to help find the bug, please say so.  (Will be out on work/mission next 2 days)

DonLakeFlyer, 2017-02-06:
I have a Solo so I can take look at this when I finally get home next week.

jaxxzer, 2017-02-13:
@DonLakeFlyer Did you try this? I have heard from one other person running windows on an [intel atom cherry-trail box](https://www.amazon.com/gp/product/B01FS6SIUE/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) about laggy video after the video recording was added. They don't have any issues with 3.1.2.

DonLakeFlyer, 2017-02-14:
Trying to repro but I can't seem to get any video to my mac box at all now with any QGC version old or new. 3DR Solo app works fine on phone so I know video is working. Likely something screwed up with my setup not QGC. Will need to do so more hacking to figure.

DonLakeFlyer, 2017-02-19:
I'm getting really laggy video even using 3.1.3 Stable. So not sure what is going on there. This is with a Typhoon. I can't get video through from my Solo any more on any of my devices.

jaxxzer, 2017-04-12:
I've been able to reproduce this, or something like this under certain circumstances. I'll post more details when I've narrowed it down a bit more.

jaxxzer, 2017-04-18:
I've been testing with a Raspberry Pi camera and a USB H264 camera connected to a Raspberry Pi 3. The RPi is connected to the topside computer with a network cable (not wireless), which may be necessary to reproduce this consistently. This issue occurs on master (doesn't affect recorded video in any case), and 3.1.3. I was never able to reproduce this issue when using the Raspberry Pi Camera, but the issue occurred consistently when using the USB camera. To reproduce the issue, plug in your cables, start the stream, and open up QGC. Everything goes according to plan. Then briefly hotplug the cable, and now the video window updates at about 1fps. Close/reopen QGC, video is fine again.

The same steps can be taken to demonstrate this issue using this command line pipeline to receive:

```
gst-launch-1.0 -e -v udpsrc port=5600 ! application/x-rtp, payload=96 ! rtph264depay ! avdec_h264 ! autovideosink
```

This outputs the following warning each time a frame appears after the issue has occured:
```
WARNING: from element /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage: A lot of buffers are being dropped.
Additional debug info:
gstbasesink.c(2791): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage:
There may be a timestamping problem, or this computer is too slow.
```
 
Adding the option 'sync = false' to the autovideosink here (as well as in QGC) removes the issue. Adding an rtpjitterbuffer also fixes it, but that introduced significant latency for me.

The sending pipeline, and caps with the raspberry Pi camera (never any issue here):
```
raspivid --timeout 0 --output - | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=10 pt=96 ! udpsink host=192.168.2.1 port=5600

/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = "video/x-h264\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ framerate\=\(fraction\)0/1\,\ parsed\=\(boolean\)true\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au\,\ profile\=\(string\)high\,\ level\=\(string\)4\,\ codec_data\=\(buffer\)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0"
```

The sending pipeline, and caps with the USB camera (always an issue here):
```
gst-launch-1.0 -v v4l2src device=/dev/video1 do-timestamp=true ! queue ! h264parse ! rtph264pay config-interval=10 pt=96 ! udpsink host=192.168.2.1 port=5600

/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ width\=\(int\)1920\,\ height\=\(int\)1080\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ colorimetry\=\(string\)2:4:7:1\,\ framerate\=\(fraction\)30/1"
```

Could the difference in stream format be the cause of the issue here (avc vs byte-stream)?

@NaterGator Any ideas here?

PS, I can reproduce the issue as described above on an i5 laptop under ubuntu 14.04, and on an i7 laptop under ubuntu 14.04 and windows 10. However, I was not able to reproduce the issue on my (much faster) desktop pc under ubuntu 16.04 or windows 10.

NaterGator, 2017-04-18:
@jaxxzer you're hitting the nail on the head with sync=false on the video sink. We absolutely do not care about global sync (we're not trying to present video+audio at the same time), we just need to present live video as quickly as possible. My advice would be to insert a queue element between the decoder and the videosink, and set sync=false on the video sink. May be worthwhile to inhibit qos messages on the bus too with qos=false on relevant elements.

This is most likely a clock source problem; when the source of data on the pipeline disappears or restarts the timestamps get skewed and the pipeline on our end is not managing this effectively. Managing clocks in gstreamer is nuanced and we have a complicated situation.

jaxxzer, 2017-04-18:
Thanks a lot @NaterGator! I will fix this soon.

jaxxzer, 2017-04-19:
@NaterGator What is the reasoning behind an additional queue?

jaxxzer, 2017-04-19:
@AndKe Can you test this https://github.com/mavlink/qgroundcontrol/pull/5023 and see if the issue is resolved?

NaterGator, 2017-04-20:
@jaxxzer a single queue offers the most benefit, but the en/decoder is the most expensive operation in the pipeline and it makes sense to isolate things up/downstream of it in their own threading contexts. In the case of the videosink you don't want the decoder blocking while the video sink tries to flip buffers or manage the display context. Just pass a decoded frame to the thread management system and immediately  go back to decoding video. 

Most systems these days have more than 2 cores to utilize, but as always this warrants testing on individual systems to tune performance.

AndKe, 2017-04-20:
@jaxxzer 
I cloned your https://github.com/jaxxzer/qgroundcontrol/tree/video-sink-sync  
built it, it did not display video from solo at all, no errors, just no video.  - three restarts of the application did not help.

QGC compiled from HEAD few days ago worked with just some delay every time.  (not the huge delay like observed before)    - unable to test with current HEAD due to https://github.com/mavlink/qgroundcontrol/issues/5025

(My office location is very packed with wifi, so it may influence the reproducibility, which is kind of good for testing of robustness.)

jaxxzer, 2017-04-20:
@AndKe What about the current daily? #5023 has been merged.

AndKe, 2017-04-20:
unable to make it,
same
```
../src/MissionManager/MissionManager.cc: In member function ‘void
MissionManager::_removeAllWorker()’:
../src/MissionManager/MissionManager.cc:917:45: error:
‘MAV_MISSION_TYPE_MISSION’ was not declared in this scope
                                             MAV_MISSION_TYPE_MISSION);
                                             ^~~~~~~~~~~~~~~~~~~~~~~~
Makefile:79438: recipe for target 'obj/MissionManager.o' failed

```

I did pull HEAD, make clean , even qmake , still same problem..

NaterGator, 2017-04-20:
@andke did you try git submodule update

DonLakeFlyer, 2017-04-20:
> it did not display video from solo at all

I ran into this as well. For me I think it was switching to a UVC device on my iPad instead. Hence it was shsowing the wrong video stream. Solo video doesn't start right away since there is a handshake that happens after startup. Just guessing now. But if that give anyone any ideas. I'll look into Solo video at some point.

AndKe, 2017-04-20:
@NaterGator yes, that was the embarrassing error, thanks.

DonLakeFlyer, 2017-05-25:
Is this still a problem. I have heard of any video problems from anyone in a while.

AndKe, 2017-05-27:
Just checked, were unable to reproduce it - looks good.

closed 2017-05-27 by AndKe

DonLakeFlyer, 2017-05-27:
Thanks a ton for verifying.