mavlink/qgroundcontrol #4516: ulog format name

Stifael, 2017-02-06:
Download via QGC of the new px4 logging format (ulog) is still appended with px4log instead of ulog.

DonLakeFlyer, 2017-04-14:
This is controller by this which needs to be put back into the firmare: https://github.com/mavlink/qgroundcontrol/issues/4971

closed 2017-04-14 by DonLakeFlyer

bellahduan, 2017-10-10:
@DonLakeFlyer hi , the log for QGC is .px4log , but it can not use to analysis . Is there any way to solve the problem?
thank u so much !

DonLakeFlyer, 2017-10-11:
The old log format is .px4log, the new format is .ulog.

bellahduan, 2017-10-11:
@DonLakeFlyer 
thank u so much . but I downloaded the log file using QGC is a.px4log, but it does not working to analyze . and I downloaded the log from SD card is a.ulg , but the same is not working well . can u, please help me to solve the problem ? 
thank u so much .

lucadelu, 2019-11-06:
Hi,

I have similar problem, but when I download the log file (a date file with .bin suffix) I'm not able to read it from any analyzer software (I tried with pyulog and the web one).
I'm using version qgroundcontrol 3.5.4.

In the case I can share a log file.

DonLakeFlyer, 2019-11-08:
> I have similar problem, but when I download the log file (a date file with .bin suffix) I'm not able to read it from any analyzer software (I tried with pyulog and the web one).

Either you are using a really old version of px4 firmware, or you are using ArduPilot firmware. Either way, pyulog won't be able to read the file.

lucadelu, 2019-11-13:
@DonLakeFlyer thanks I have ArduPilot firmware, I forgot it