mavlink/qgroundcontrol #4520: Use radio as joystick - no flight modes

ChristophTobler, 2017-02-06:
I have an FrSKY Taranis and I can use the normal 4 channels (throttle, yaw, roll, pitch) in QGC as a joystick when connected to my desktop. However I can not set the flight modes since it has no buttons. I can see more channels on the ```Axis Monitor``` but I can't use them for the flight modes. Or can I?
This would be very useful since you could fly a Bebop for example with a real controller and not a joystick.

DonLakeFlyer, 2017-02-07:
I think there are people who have figured out how to do this. I'd ask on Gitter to see if someone already has a way.

tubeme, 2017-02-13:
It depends on the firmware you are using. Before it was not possible because the driver for the Taranis Joystick did not support more than first 4 channels for control only. Later the driver was upgraded so we could use the the Taranis connected as a joystick in the simulators and use 8 channels. But I never tried it in recent versions of QGC. 

The easiest way to check is to see in parameters in the joystick menu to see what is recognized as input during the calibration procedure. If there you see only 4 channels then this is the only visible from QGC.

If you have older firmware of the opentx then you can try newer versions. There is some development for Joystick for the 2.2 version i saw in the git.

hamishwillee, 2017-07-04:
FYI I just tried the latest opentx firmware. QGC shows the 8 buttons as selectable, but moving buttons etc on the controller doesn't show anything. This is current stable QGC 3.1

mainframe, 2019-09-08:
I have opentx 2.2.3 on x9D Plus SE and buttons do work with qgroundcontrol (controlling flight modes) - but you need to configure your taranis switches properly for the model - and figuring that out is not so straightforward unfortunately.

Basicly it requires:

1. defining model button inputs on taranis (use logical switches for more complex flightmode changing setups)
2. defining model mixes on taranis channels 9-32 using previously defined inputs as sources. CH1-8 are analog axes and cannot be used for joystick buttons. 
3. mapping buttons 0-23 in QGC Joystick setup to flightmode commands for example (button0 is on CH9, button1 is on CH10, etc.). You should also see some activity on Joystick screen on button0-23 channels - when flipping defined switches on Taranis.

brunweil, 2021-03-08:
To my knowledge "Mainframe" your answer is the only explanation on the whole internet concerning this problem! congratulations!

closed 2026-04-28 by DonLakeFlyer