mavlink/qgroundcontrol #4523: Issue with same system id as QGroundControl

abencomo, 2017-02-07:
I recently installed [QGC v3.1.2](https://github.com/mavlink/qgroundcontrol/releases/tag/) and used it to upload the [PX4 v1.6.0rc1](https://github.com/PX4/Firmware/releases/tag/v1.6.0rc1) into the [Pixhawk](https://store.mrobotics.io/Genuine-PixHawk-Flight-Controller-p/mro-pixhawk1-minkit-mr.htm). Then today, I installed QGC v3.1.2 in a different machine and when I tried to connect to the Pixhawk, I got this warning message:

"[Warning: A vehicle is using the same system id as QGroundControl: 255](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/MultiVehicleManager.cc#L86)"

Although it is a warning, it still didn't allow me to access the parameters list.  I went back to the original laptop I used to upload the firmware; and I got the same message.  Does any know what could have caused this or how can it be fixed or at least prevent it from happening again?

DonLakeFlyer, 2017-02-07:
Are you running a companion computer which has a mavlink id of 255? If so, change the id for the companion computer. Or use a daily build which disregards companion computers.

abencomo, 2017-02-08:
Yes, I'm using a companion computer with ROS and MAVROS.  Where do I change the mavlink id that you are referring to?  And, where are the QGC daily build located?

samacumen, 2017-02-10:
Usually applications (like qgroundcontrol) assume a default MAVLINK_ID - which can be changed in the settings. It is very much possible that MAVROS may have this id hardcoded (not sure) - but you certainly should be able to change it. Look for it in the code, also have a look at [this issue](https://github.com/mavlink/mavros/pull/256).
:

DonLakeFlyer, 2017-02-13:
Daily build links at the bottom: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html

abencomo, 2017-02-13:
That did it.  Thanks everyone.

closed 2017-02-13 by abencomo