mavlink/qgroundcontrol #4528: Heli parameters with Multicopter frame - ArduPilot 3.5dev

lvale, 2017-02-08:
I've closed the issue on ArduPilot, because the problem appears to be with QGC

https://github.com/ArduPilot/ardupilot/issues/5690

DonLakeFlyer, 2017-02-14:
Can you detail out some repro steps. I'm not getting it.

DonLakeFlyer, 2017-02-23:
@lvale ?

lvale, 2017-02-23:
Managed to sort things on latest master. The refreshing of parameters when changing frames was the "bad guy". 
Sorry for not closing sooner.

DonLakeFlyer, 2017-02-23:
To me those frame params should have a "reboot required" on their meta data (firmware change). All sort of strange things happen when you change them on the fly.

lvale, 2017-02-27:
@DonLakeFlyer Just retested on the latest master and the issue of changing frame is there:

better see the video https://www.youtube.com/watch?v=RKxgJ4YO_Wo

DonLakeFlyer, 2017-04-13:
This is because of the firmware change in ArduCopter where the newer rc versions doesn't have separate firmwares for each vehicle type. Once ArduCopter releases 3.5 this will go away since all versions will use this new scheme. Has 3.5 been released yet?

lvale, 2017-04-14:
3.5 not yet, still RC.. 
I believe the behaviour is that when you change something on the bottom dropdown box, it will reset the top dropdown box to the first element.

DonLakeFlyer, 2017-04-14:
The two lists are different so you can't really preserve selection when you switch. Once the lists are the same I can preserver selection.

DonLakeFlyer, 2017-05-08:
@lvale Is 3.5 out yet?

lvale, 2017-05-08:
@DonLakeFlyer Still at RC5

DonLakeFlyer, 2017-05-24:
I looked at the code for this yet again. Trying to maintain the selection vehicle firmware type when you change firmware type (stable, beta, ...) is a pile of changes. All to change something which will go away once 3.5 goes out. So I'm closing this since it's not going to get changed.

closed 2017-05-24 by DonLakeFlyer