mavlink/qgroundcontrol #4530: Communication lost with Vehicle (over Serial)

samacumen, 2017-02-08:
When QGC is connected to Pixhawk via Serial, I sometimes find that it says "Communication lost". I need to plug the serial cable back in to get the mavlink flowing between the two. I have tested with different USB cables, the quality of cable is fine.

QObject::connect: invalid null parameter seems to be the problem somewhere trying to connect to QML object, which breaks and stops the flow of mavlink.

Starting /home/sam/build-qgroundcontrol-Desktop_Qt_5_5_1_GCC_64bit-Debug/debug/QGroundControl...
Settings location "/home/sam/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/sam/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
Map Cache in: "/home/sam/.cache/QGCMapCache300" / "qgcMapCache.db"
**QObject::connect: invalid null parameter
QObject::connect: invalid null parameter
QObject::connect: invalid null parameter
QObject::connect: invalid null parameter**
LinkManager mavlinkStatus:channel:flags 0x1203408 1 2
LinkManager mavlinkStatus:channel:flags 0x1203430 2 2
Temp log "/tmp/FlightData686202.mavlink"
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 2
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 3
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 4

[image]

dogmaphobic, 2017-02-08:
> QObject::connect: invalid null parameter seems to be the problem somewhere trying to connect to QML object, which breaks and stops the flow of mavlink.

That seems like a signal/slot connection passing a NULL pointer.  Is this using the latest master? I haven't seen this before.

samacumen, 2017-02-08:
We are using branch v3.1.1 tag of qGroundcontrol. So, looks quite recent.
It happens mostly when qgc is started with a usb cable plugged in, or after
plugging in a cable on started usb. I thought to report it right away.

samacumen, 2017-02-08:
When I plug back the usb in the vehicle instance is created (again). Somehow it gets lost, as you said signal/slot NULL pointer looks like the possibility. Would be interested if you could reproduce the same at your end. Sometimes it happens quite regularly.

1) Connect the Pixhawk
2) Start qgc (launched in release mode via qtcreator)
3) In principle, we should get the green progress bar and Vehicle connected. But sometimes, it immediately says (after 3 second timeout) of vehicle connection, that the Connection is lost.

samacumen, 2017-02-09:
Found this:
qrc:///qml/QGroundControl/FlightMap/CenterMapDropButton.qml:44: TypeError: Cannot call method 'get' of null
qrc:///qml/QGroundControl/Controls/RallyPointEditorHeader.qml:34:9: QML Rectangle: Binding loop detected for property "height"
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 2
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 3
Vehicle::_sendMavCommandAgain retrying command:_mavCommandRetryCount 520 4

Maybe, this is the cause?

DonLakeFlyer, 2017-02-21:
> QObject::connect: invalid null parameter seems to be the problem

This has been fixed in daily builds.

> Found this:
> qrc:///qml/QGroundControl/FlightMap/CenterMapDropButton.qml:44: TypeError: Cannot call method 'get' of null
> qrc:///qml/QGroundControl/Controls/RallyPointEditorHeader.qml:34:9: QML [...]

None of those matter.

samacumen, 2017-02-21:
Get the same message as you showed. Object Null is not found, I tried it a number of times on daily build. Closing this issue! Thanks.

closed 2017-02-21 by samacumen