mavlink/qgroundcontrol #4532: Qgroundcontrol 3.1.2 video corruption

svpcom, 2017-02-09:
I've got periodic video corruption (see screenshots attached) with test video stream on localhost.
Host system is a Linux x86-64.

Video source:  ```gst-launch-1.0 videotestsrc ! videoconvert ! clockoverlay ! x264enc tune=zerolatency key-int-max=32 ! video/x-h264,width=800,height=600,framerate=30/1 ! rtph264pay ! udpsink host=localhost port=5600```

If i receive video using gstreamer (without qgroundcontrol) i haven't got any corruptions:  ```gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! clockoverlay valignment=bottom ! autovideosink sync=true```

CPU is not saturated - i.e. it seems not a frame drop.

[image]

[image]

svpcom, 2017-02-21:
@DonLakeFlyer  Do you need any additional details to reproduce it? How difficult to fix this?

DonLakeFlyer, 2017-02-21:
Haven't look at it. @dogmaphobic Will have to look.

dogmaphobic, 2017-02-22:
The screen grab above doesn't have the recording button. That would indicate it's an older version? I'm having some issues with RTSP (unrelated to this) but I just tried UDP and it's working fine.

[image]

svpcom, 2017-02-22:
@dogmaphobic I've retest it with latest master (a56a153d2b8a9dede579058b2c59fd71428a6805) and the issue is fixed there.  But it exists in latest stable release 3.1.3

dogmaphobic, 2017-02-22:
Is it then safe to close this issue?

svpcom, 2017-02-22:
Yes, but I think we need to identify and backport fix to stable branch.

DonLakeFlyer, 2017-02-22:
Not sure what would have changed from stable to master for video? I think only the recording stuff was added. The mainline video code didn't change I think.

NaterGator, 2017-02-22:
I am surprised we have latency set to 0 for RTSP sources and no jitterbuffer for udp sources. In my experience the network is very very rarely good enough to deliver the packets in order and without issues... even a buffer of 1 frame period is usually long enough.

DonLakeFlyer, 2017-05-08:
Closing since daily is working. Not going to back port to stable.

closed 2017-05-08 by DonLakeFlyer