mavlink/qgroundcontrol #4556: QGC crashes during running HIL (on network error/ on px4 disconnection)

samacumen, 2017-02-16:
To reproduce this issue, 
1) Open QGC and connect with Pixhawk over USB
2) Open the 'HIL Config' widget
3)  Start HIL (with standard settings connection to X-Plane)
4) See that connection to simulator (X-Plane or any other simulator is established)
5) Now, disconnect Pixhawk (plug off USB). The QGC should crash.

_**Reason**: LinkManager thinks it is still connected to the X-Plane on the UDP link when it has already been (abruptly) disconnected (The knowhow of "Link to X-Plane" is coupled to the Vehicle class - this is questionable!). Apart from maintaining the correct state, the HIL must also come out of the thread execution loop [QGCXPlaneLink::run()] - it doesn't. No one makes the "connectState" flag to false, the loop is stuck infinitely._

I came up with a solution as below (commented). Tested to work well.
```
/**
 * @brief Check if connection is active.
 *
 * @return True if link is connected, false otherwise.
 **/
bool QGCXPlaneLink::isConnected()
{
    // We make sure that when the link is detected as disconnected by the LinkManager, QGCXPlane
    // also detects it at the earliest and sets the connectState flag to false. Ideally the UDP
    // connected state must be other than 'BoundState or ConnectedState' when the vehicle has left.
    // However, it is always seen to be in 'BoundState' inspite of network error/broken link and
    // the connection thread is held in the wait state. A practical solution is to detect if the
    // vehicle is disconnected and set the connectState flag to false. This is a fix for QGC crashing
    // when HIL has been started and Pixhawk is disconnected manually.
    if (!_vehicle || (_vehicle && !_vehicle->active()) )
        connectState = false;

    return connectState;
}

```

DonLakeFlyer, 2017-02-21:
I tried latest master and it's working fine for me. Where exactly is it crashing?

samacumen, 2017-02-21:
Thanks for checking it out. I just verified this on the latest build master, but unable to reproduce it this time. I will anyhow close this issue. On a side note, when the user disconnect's by pressing the disconnect button, the streaming text "Received from XPlane at xHz" should change to something like "Disconnected X-Plane". Helps in indication.

[image]

closed 2017-02-21 by samacumen