mavlink/qgroundcontrol #4560: "Save Flight Data Log" ,provide filename for organized storage.

AndKe, 2017-02-16:
Before exiting user is asked to save tlog, or not.
I wish there were options like  "always save", "save if vehicle was armed",  or "ask user".

apmplanner2 have a great system for organizing logs, that I hope QGC can adopt:
(Please let's all have one uniform way to store logs)

"vehicle type directory"/"yyyy-mm-dd HH-mm-ss".tlog

resulting in plane/2017-02-15 11-09-15.tlog

DonLakeFlyer, 2017-02-16:
Coming soon...

AndKe, 2017-02-16:
fantastic, please allow to select default directory for this structure as
well I am always syncing logs across computers (so they are all in place
regardless of which laptop was used (and soon regardless of GCS used))

closed 2017-03-15 by DonLakeFlyer