mavlink/qgroundcontrol #4561: Please allow (much) longer trailing track

AndKe, 2017-02-16:
the track disappears, only a certain length is visible, please make it possible to select a much longer trail.

DonLakeFlyer, 2017-02-16:
FYI: They used to be infinite. The longer vehicle tracks ended being a perf problem on mobile. So I set it to a fixed size. But didn't spend the time to add a settings for it.

I'll add a setting back for it.

AndKe, 2017-02-16:
fantastic, another way to reduce, but keep enough on mobile, would be to reduce by just removing two of thee (or more) instead of all.

DonLakeFlyer, 2017-02-16:
Are you saying that the whole track is disappearing. The way it should be working is that there is a max number of points. When you hit that max, points are removed from the start of the track so you can always see the end of the track.

AndKe, 2017-02-17:
No, sorry for misformulating that, it works as designed, (last points are
lemoved).
I am just asking for longer track, by having more points, and/or by
reducing point density.

DonLakeFlyer, 2017-02-17:
Got it thanks. Just wanted to make sure.

DonLakeFlyer, 2017-05-08:
Desktop builds are now back to infinite length. Mobile still limits to three minutes of data. Will need fancy mechanism for mobile in next release.

AndKe, 2017-05-08:
nice !

DonLakeFlyer, 2017-12-30:
Going to leave this like it is for now

closed 2017-12-30 by DonLakeFlyer