mavlink/qgroundcontrol #4562: Climb out altitude for MAV_CMD_DO_GO_AROUND hardcoded

dkschrad, 2017-02-16:
The climb out altitude for MAV_CMD_DO_GO_AROUND is currently hardcoded to 50 m.  Something needs to be implemented to properly set that altitude.

DonLakeFlyer, 2017-02-16:
This is use the altitude slider from the guided bar with an appropriate default on the slider.

DonLakeFlyer, 2017-05-08:
I can't find a reference to this command anywhere in the code. So no idea what this is.

dagar, 2017-05-08:
MAV_CMD_DO_GO_AROUND is what we send to abort a landing. It's 1 and only param is an altitude.

DonLakeFlyer, 2017-05-08:
Ah, I wasn't look for the mavlink method correctly. Duh

DonLakeFlyer, 2017-05-08:
I assume this isn't critical for 3.2 stable?

dagar, 2017-05-08:
I'm not sure if we actually need it at all. Aborting a FW landing needs to happy quickly, but after it's safely climbing out you could adjust altitude the normal way.

DonLakeFlyer, 2017-05-08:
That makes sense. You don't want to be confused by extra stuff. We can look at this again in 3.3.

DonLakeFlyer, 2018-06-20:
Going to live with this until more complaints

closed 2018-06-20 by DonLakeFlyer