mavlink/qgroundcontrol #4564: disconnecting a link when multiple links are connected to one vehicle

mhugentobler, 2017-02-17:
I connect to my vehicle via a Serial and a UDP link. When both links are connected and I want to disconnect only the Serial link by clicking "disconnect" in the link settings, often, tough not every time, I cannot reconnect this link, because the "connect" button is not clickable. The only thing that helps is to wait for the "communication lost" message and then click "disconnect" in the main tool bar. After that obviously both of my links are disconnected and not only the Serial one. 

In v3.0 this was not an issue. Whenever you would disconnect the Serial link while the UDP link was still connected, you could reconnect at any time, while having persistent connection via UDP.
Since v3.1 this does not work reliably anymore.

DonLakeFlyer, 2017-02-17:
Can you try a daily build? There have been a number of fixes around this which have been made since the last stable.

mhugentobler, 2017-02-20:
Just tried it with current master (as of now). The problem is still there: connect serial -> connect udp, then disconnect and reconnect serial a couple times. Usually at the second disconnection of the serial link the problem described above occurs.

DonLakeFlyer, 2017-02-20:
Sorry I misread your issue. Thought it was multi-vehicle.

DonLakeFlyer, 2017-03-01:
You should turn off all auto-connect from Settings and try this again using manual links only. Most likely auto-connect getting in the way.

mhugentobler, 2017-03-01:
I have never used auto-connect before (always all disabled), so I suppose the issue occurs due to something else.

DonLakeFlyer, 2017-05-26:
I understand what is going on here now. It's a big change to fix so it will go in first for 3.3.

DonLakeFlyer, 2018-01-15:
@mhugentobler Some of the connect/disconnect semantics have been reworked. Is this still happening with latest daily builds?

mhugentobler, 2018-01-16:
Since I am no longer working with QGC I cannot comment on this. Maybe @acfloria knows more

acfloria, 2018-02-02:
@DonLakeFlyer I checked it today with two serial links.

1. I disabled autoconnect.

2. Open a serial connection to ttyUSB0

3. Open a serial connection to ttyACM0. 

4. Try to disconnect ttyUSB0. On the link screen it still shows as connected but after some time I get the `Communication lost` message, shortly followed by a `Communication regain` notification. Though even after this notification I cannot for example upload a mission. I can not edit ttyUSB0 as it still is shown as connected. 

5. After disconnecting ttyACM0 both connections are shown as disconnected. No vehicle connection is present and I can edit the connections.

If I first close the 2nd link I connected I can edit and reconnect it.

DonLakeFlyer, 2018-02-02:
Ok, I'll take a look at this again.

DonLakeFlyer, 2018-02-02:
There is a problem with shared link refs. Someone is hold an additional ref on the first link. That goes away, when you disconnect the second link.

DonLakeFlyer, 2018-06-20:
I believe fixed now

closed 2018-06-20 by DonLakeFlyer