mavlink/qgroundcontrol #4569: crash switch to Airframe

zenetio, 2017-02-18:
QGC 3.1.3, Windows 10
When I switch from one option to Airframe, the QGC crashes and I need start it again. It was happening with QGC 3.1.2 and continue with 3.1.3

DonLakeFlyer, 2017-02-18:
What OS, Firmware and firmware version?

Also do you mean when you switch to the Setup/Airframe page from any other Setup page?

zenetio, 2017-02-19:
I am using Windows 10, QGC 3.1.3 and FW v1.5.5
The crash happens when I switch from one icon to Airframe. For instance, from Sensors to Airframe or from Firmware to Airframe

DonLakeFlyer, 2017-02-20:
Can you save your parameters to a file and attach here?

Stifael, 2017-02-22:
I am not sure if this issue is related tho the issue that I encountered:
The vehicle was hovering close to the groundstation, and everytime I opened the analyzer and observed some value, QGC crashed after around 30s

zenetio, 2017-02-22:
After save parameters, I switched from Joystick to Airframe and got crash.
[px4.params.txt](https://github.com/mavlink/qgroundcontrol/files/793692/px4.params.txt)

DonLakeFlyer, 2017-04-13:
Can you try the latest daily build?

zenetio, 2017-04-13:
Sure

DonLakeFlyer, 2017-05-07:
@zenetio Any update?