mavlink/qgroundcontrol #4572: Mission Planning: last used waypoint altitude function should ignore special cases such as LAND

Antiheavy, 2017-02-18:
Special case waypoints such as LAND should ignore the last used waypoint altitude and these waypoints should also be ignored by other waypoints when determining last used altitude.

Currently if waypoints are placed numerically after a LAND or inserted earlier into the mission they default to the land altitude (usually set to 0 altitude, although at time of writing https://github.com/mavlink/qgroundcontrol/issues/4522 is still open) - this is probably bad as you wouldn't want to "fly" into the ground.  LAND waypoints should not follow the last used altitude and other waypoints should not follow Land waypoints' altitude.

This should be extended to ignore landing patterns and other complex waypoint (or mission item) types that have multiple elements at different altitudes.

Also check the behavior of: Takeoff, Loiter, VTOL takeoff and transition, VTOL transition and land.  These currently do not follow the last used altitude as they are hard coded to 50 m.  However, waypoints placed after these follow the altitude of these waypoints (with the exception of Takeoff waypoints) .  Takeoff is the only waypoint type that seems to ignore this logic both ways - it does not follow last used altitude and other waypoints do not follow its altitude.  I'm not sure if this is actually the desired behavior for Takeoff waypoint, but I think it is fine.  In fact, this is how the Land waypoint should work too.  For consistency, if a waypoint ignores the last used altitude then its altitude(s) should also be ignored.

(i'm using PX4 and Windows QGC v3.1.3)

DonLakeFlyer, 2017-02-20:
I'm also thinking the current logic to determine last used altitude is wrong with respect to the fact that it always pulls the last altitude in the list even if you are inserting and item in the middle of the mission. I think it should pull the altitude to use by only walking back from the insert point.

Antiheavy, 2017-02-20:
I guess walking back from the insert point would work for waypoints placed at the end of a mission list too.  As long as the walking back logic skips over special case waypoints such as Takeoff, Land, Patterns, and maybe others.

closed 2017-02-21 by DonLakeFlyer