mavlink/qgroundcontrol #4579: Low Resolution Map

KeeyenPang, 2017-02-20:
As per suggested, I raise the issue here. The original post was on ArduPilot [http://discuss.ardupilot.org/t/low-resolution-map/14920](url)

I've tried install in three difference computer, running on Win 8 and Win 10 with the same result.

dogmaphobic, 2017-02-20:
The link above returns a 404. What is this about?

DonLakeFlyer, 2017-02-20:
@KeeyenPang Can you repost you images and details here instead of linking to ArduPilot forum?

KeeyenPang, 2017-02-23:
Dear Don

I've found the culprit, it is Bing Map issue. Once I change to Google, all went well. For the benefit of those who want to use Bing map, I re post the map here. Thanks

Keeyen
[image]

closed 2017-02-23 by DonLakeFlyer