mavlink/qgroundcontrol #4593: Android daily builds not working

DonLakeFlyer, 2017-02-21:
I see this:
[image]

I'm not quite sure how the daily side works so not sure how to fix it myself.

dagar, 2017-02-22:
> Assigned #4593 <https://github.com/mavlink/qgroundcontrol/issues/4593> to
> @dagar <https://github.com/dagar>.

I'll look at this tonight. Need to drop something in master for versioning.
Maybe a dev tag? `v3.2-dev`

DonLakeFlyer, 2017-02-22:
Thanks Daniel.

dagar, 2017-02-23:
Hmm, might be easier to have a completely separate play store entry for QGC Beta. 
Any objection?
It would be nice if the dev version had a slightly different icon, but that's optional.
[image]

The upside is that we'll be able to install and use stable and dev side by side on the same device.

DonLakeFlyer, 2017-02-23:
Perfectly fine by me. If you update can you update the docs as well which has links to the android beta stuff.

DonLakeFlyer, 2017-02-23:
You could ask the icon man: @dogmaphobic for a beta icon for the site.

dagar, 2017-03-05:
Going off the grid for a few days, and I don't think I'll have this working before I leave. I'm going to create a new Play Store entry for "QGroundControl Beta" and it needs a few more things than I initially realized. Changes in the build system to update the package name, deploy permissions, etc.
@dogmaphobic could you do a quick icon like the chrome example above?

In the meantime Android users can directly install the master build. https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.apk

dagar, 2017-04-09:
https://github.com/mavlink/qgroundcontrol/pull/4947

closed 2017-04-13 by DonLakeFlyer