mavlink/qgroundcontrol #4596: With QGC running, closing and reopening the laptop screen greys out QGC screen/tab icons

samacumen, 2017-02-21:
_Test with:_
Ubuntu 16.04
Build: Latest QGC master build (21 Feb, 2017)

We often go out to the field for testing by leaving the QGC in running state. We obviously need to close the laptop while travelling.  When we arrive, we open the laptop to find something like below. Not sure if it is the qt or Ubuntu issue but have a look anyways: 

[image]
[image]

NaterGator, 2017-02-21:
This screams buggy OpenGL driver. Are your GPU drivers up to date? Can you get newer drivers from your GPU vendor?

DonLakeFlyer, 2017-02-21:
I get that a lot from running in a VM. QGC isn't in control of it.

samacumen, 2017-02-21:
I am not running it on a VM. I have only 1 OS installed on my system i.e.
Ubuntu. Maybe a Linux library issue or something? To confirm, would be a
good idea to check it on APMPlanner2.0,  which is also built on qt.

DonLakeFlyer, 2017-02-21:
APMPlanner and QGC user very different mechanisms to show ui. That won't tell you much.

closed 2026-10-05 by github-actions[bot]