mavlink/qgroundcontrol #4622: qgroundcontrol build error (  Error: C2440: "Initialization": Can not convert from "initializer-list" to "QMap <int, QString>" The constructor for "QMap <int, QString>" is declared as "explicit")

goodman110110110, 2017-02-25:
hi, when i try to compile the qgroundcontrol3.0.0 source code with qt5.5.1 ctreator, an error appear.

 Error: C2440: "Initialization": Can not convert from "initializer-list" to "QMap <int, QString>" The constructor for "QMap <int, QString>" is declared as "explicit"

i followed the readme.md to download the qt version 5.5.1 and to build the code.

please help me . Thankyou. 
[image]

DonLakeFlyer, 2017-02-25:
Are you using VS 2015 to build. QGC only support VS 2013 at the moment.

goodman110110110, 2017-02-26:
I used vs2013， I followed the readme.md to complie the code.
but it can not work.

18911847946, 2017-02-26:
i have met the same trouble

18911847946, 2017-02-26:
[image]

DonLakeFlyer, 2017-02-27:
Do you have the same problem when you build latest master?

goodman110110110, 2017-03-02:
yes, we have the same problem. there have many mistakes when qt creator was used.

goodman110110110, 2017-03-02:
Yes, we have the same problem, following the readme.md to build the latest
code, but the qt creator comes out many mistakes. please help me!
Thank you!

booo, 2022-11-01:
Does not concern newer versions of the software. Closing therefore.

closed 2022-11-01 by booo