mavlink/qgroundcontrol #4633: Qt design wont open .qml files - Right or wrong?

crashmatt, 2017-02-26:
I am attempting to open .qml files with design.  Is this possible?  If not then please tell me to stop it.  If possible then there is a problem.

Everything builds ok but...
When browsing qml files it can't find the imports below.  When launching editing in design mode it reports that it can't find these imports.

import QGroundControl.FactSystem    1.0
import QGroundControl.FactControls  1.0
import QGroundControl.Controls      1.0
import QGroundControl.ScreenTools   1.0

Added this to the project in hope of a fix:

QML_IMPORT_PATH += src
QML_IMPORT_PATH += src/QmlControls
QML_IMPORT_PATH += src/FactSystem
QML_IMPORT_PATH += src/FactSystem/FactControls

Tested on Qt 5.5.1 and 5.8.   5.5.1 just crashes everything. 5.8 handles it and closes the editor gracefully.   The .ui forms open and edit ok on both versions.

In Qt5.5.1 I get lots of these warnings when opening the project:

Warnings while parsing QML type information of /home/matt/Qt5.5.1/5.5/gcc_64/qml/QtLocation:
/home/matt/Qt5.5.1/5.5/gcc_64/qml/QtLocation/plugins.qmltypes:1:24: Reading only version 1.1 parts.
/home/matt/Qt5.5.1/5.5/gcc_64/qml/QtLocation/plugins.qmltypes:10:5: Expected only Component and ModuleApi object definitions.

In Qt5.8 with Qt5.5.1 kit I don't get any of those messages.

Are there any special conditions or paths I need to set to get this to work? Should it work? Am I expecting something that simply doesn't work.

/Matt

DonLakeFlyer, 2017-02-26:
Stop. You have to edit Qml by hand. Two problems:
1) Most QGC qml uses a C++ controller which pushed values into the Qml so designer wont' work
2) Most QGC Qml is way to complex to edit in the designer

crashmatt, 2017-02-26:
Running ubuntu 16.04

closed 2017-02-26 by DonLakeFlyer

crashmatt, 2017-02-26:
ok, I was expecting too much.