mavlink/qgroundcontrol #4635: FixedWing Landing Pattern: Remaining work

DonLakeFlyer, 2017-02-26:
Remaining work:
* Click to set current item
* Persistence
* Mission stats calculations
* More work on visuals

Questions:
* Exit cross-track MAV_CMD_NAV_LOITER_TO_ALT command is set to 0. Does this mean the vehicle will exit the loiter on a direct line from loiter to land waypoints center?
* Need feedback on default values

Current state:
* Ready for SITL testing to gather feedback before I finish out the details.

DonLakeFlyer, 2017-02-26:
@antiheavy FYI...

NaterGator, 2017-02-28:
Exiting the loiter on a tangent point is virtually all I've seen in use because it provides for the most predictable transition to landing approaches. Also computing the glide slope from the breakout point to landing coordinate is helpful to determine if you are within the aerodynamic margins of your craft. 

I've succeeded in getting NaNs on the landing point distance and heading; I think it is happening when distance == heading, but it is a bit hard to say. Around those same singular points I've also seen the "L" abbreviation disappear on the two position indicators.

Antiheavy, 2017-03-01:
copying my comments from https://github.com/mavlink/qgroundcontrol/pull/4599 to here:

I get the following error messages (yellow pop-up bar) when uploading missions with Landing Patterns:

- "Error during Mission communication with Vehicle: Vehicle returned error: Command is not supported (MAV_MISSION_UNSUPPORTED). Vehicle only has a partial list of mission items."
- "IGN MISSION_ITEM: Busy"

The Loiter-to-Alt and Land waypoints get taken by the autopilot as normal waypoints at default altitude. This is confirmed by downloading from the autopilot after attempted upload.

Otherwise looking great! Here is some additional feedback:

Default "Loiter altitude" of 40 m is good. Above most trees and local power lines in the world. Obviously pilot must adjust on their own for hilly terrain, giant trees, large structures, etc.

Default "Landing distance" should be 300 m. 40 m alt and 300 m distance provides a 7-8 degree glide slope which is reasonable for most small fixed wings. A future ability for users to set their own defaults for things like this will be nice.

"Landing heading" should be more like how runways work in aviation. The heading angle should be the direction of flight of the aircraft, not the direction the aircraft is landing from (i.e. 180 degrees opposite from how it is in this PR). (I feel there might be other places in QGC where this is "wrong", wind direction comes to mind)

(L) icon should only be used for the Landing Point to avoid confusion. Other points in the pattern should have some other identifier that shows they are part of the landing pattern, but are not the Landing Point. Suggest (L), (L1), (L2), etc working backwards from (L), or something else that easily identifies it as part of the pattern, but not the actual Landing Point. (I feel this could apply to other pattern types such Survey where S1 and S2 may more clearly indicate start and stop of a Survey)

Added bonus: it seems that after placing a Landing Pattern the waypoint tool is no longer active so accidental waypoint creation after the Land Point is now much less common! Seriously, this removes a big annoyance in an elegant way.

DonLakeFlyer, 2017-03-01:
> "Error during Mission communication with Vehicle: Vehicle returned error: Command is not supported (MAV_MISSION_UNSUPPORTED). Vehicle only has a partial list of mission items."

Fixed by #4665

DonLakeFlyer, 2017-03-01:
> it seems that after placing a Landing Pattern the waypoint tool is no longer active

Yeah, this was on purpose.

DonLakeFlyer, 2017-03-01:
@Antiheavy Rest of comments all make sense. I'll get those changed.

DonLakeFlyer, 2017-03-02:
> (L) icon should only be used for the Landing Point to avoid confusion. Other points in the pattern should have some other identifier that shows they are part of the landing pattern, but are not the Landing Point.

Fixed by #4669

DonLakeFlyer, 2017-03-02:
> Exiting the loiter on a tangent point is virtually all I've seen in use because it provides for the most predictable transition to landing approaches. Also computing the glide slope from the [...]

Is that in response to this question: "Exit cross-track MAV_CMD_NAV_LOITER_TO_ALT command is set to 0. Does this mean the vehicle will exit the loiter on a direct line from loiter to land waypoints center?"

If so, is that cross track value correct then that I'm sending?

DonLakeFlyer, 2017-03-03:
> Changing the landing point properties (distance/heading) is actually moving the loiter point.

This is on purpose since you click to set the landing point. The assumption is that is where you want to land since you just clicked there so the loiter is adjusted not the land point.

DonLakeFlyer, 2017-03-03:
Fixed:
* Landing distance defaults to 300m
* Heading is now from loiter to landing (previous: landing to loiter)

DonLakeFlyer, 2017-03-03:
Fixed:
* Correct cross track value loiter to be tangent exit for item sent to vehicle
* Visuals shows tangent flight path

DonLakeFlyer, 2017-03-05:
Completed:
* File save/load
* Mission stats

This completes the item listed here. Rest is testing/bug fixes and items in other Issues.

closed 2017-03-05 by DonLakeFlyer