mavlink/qgroundcontrol #4636: Prototype user interface for mixer tuning

crashmatt, 2017-02-26:
To test mixer tuning backend code I need a user interface to communicate with.  Some basic edit boxes and buttons to start with.  

Something that hangs off the config menu would be a good place to put it.  I think I see bits that I can copy such as the parameter editor.   Suggestions of where to get started would be welcome.

The shape of the final user interface will be open for debate.  For now I would be happy with some basics.

/Matt

DonLakeFlyer, 2017-02-26:
I'll add the framework for you to get started with. Then you can fill in the guts.

crashmatt, 2017-02-26:
Thanks Don.  Where will you put the new stuff?
I can add you as a collaborator to by fork if that helps?

DonLakeFlyer, 2017-02-26:
This is mavlnk generic support right? Not specific to a firmware?

DonLakeFlyer, 2017-02-26:
I would suggest putting through small pulls so I can review and make sure you are using the right parts of the QGC infrastructure. Also make sure to ask me any questions you need to. If you have some sort of initial mock up (pencil drawing is fine) as well as some example of how you need to talk to the vehicle then I can point you in the right direction.

crashmatt, 2017-02-26:
The target is for it to be generic and it is written in a generic way.  The plan is to have the mavlink messages in common.  I already have it running with mavproxy and ardupilot.

DonLakeFlyer, 2017-02-26:
Ok, good. The pull I put through adds Mixers to both PX4 and ArduPilot firmwares. The Mixer code is common to both. Depending on what you need you may also need to create a custom C++ controller object to drive the Qml. Depends on how you need to talk to the vehicle and whether what you need is already in there.

crashmatt, 2017-02-26:
Well that saved me a whole bunch of stress attempting to find where to connect things.
I will need to work on a custom c++ controller, a bit similar to parameters.

closed 2017-12-31 by DonLakeFlyer