mavlink/qgroundcontrol #4643: Initial loading of parameters extremely slow-ArduPilot master

lvale, 2017-02-27:
After flashing the PixRacer the initial loading of parameters takes forever and eventually QGC crashes

https://www.youtube.com/watch?v=UFfJoiQ61Bo

[crash.mavlink.zip](https://github.com/mavlink/qgroundcontrol/files/802443/crash.mavlink.zip)

DonLakeFlyer, 2017-03-02:
The problem was triggered by a change in copter which makes some parameters be missing at boot. Talked to tridge about it a week or so ago it and he changed it to delay reponse to param_request_list until all params are available. But this in turn causes QGC to timeout and drops to a slow fallback mechanism to get params. 

I need to figure out how to either not hit the fallback mechanism or to speed up the fall back mechanism.

I've never seen the crash before, but not entirely suprising since it's being forced into a strange codepath which doesn't happen often. I can't see the video to see which version of QGC you are using. I assume a new daily?

DonLakeFlyer, 2017-03-02:
Nope I was wrong. ArduPilot is just not working correctly on initial request. It sends one param and then it stops.

DonLakeFlyer, 2017-03-02:
And that wasn't it either. Just more strangeness with ArduPilot and the param protocol. I still can't get the crash though. Can you retry one this goes throuhg?

lvale, 2017-03-03:
@DonLakeFlyer As far as I can see the issue is solved (connected on USB) when we flash the firmware for the first time on a "blank" PixRacer
latest buid of QGC
https://youtu.be/RmTm06rza_0

DonLakeFlyer, 2017-03-03:
Awesome, thanks for verifying @lvale

lvale, 2017-03-06:
ok. Via WiFi I had to give up after a few minutes waiting. No parameters downloaded. 

Tested again with APM Planner and downloaded 752 parameters (ouch!!) in about a minute

Adding QGC log - tested with 

[image]

[qgc_no_parameter_download_via_wifi.txt](https://github.com/mavlink/qgroundcontrol/files/822402/qgc_no_parameter_download_via_wifi.txt)

DonLakeFlyer, 2017-03-06:
Fixed in latest master: https://github.com/mavlink/qgroundcontrol/pull/4723

lvale, 2017-03-06:
Just retested with #4723 - Still here :(
[latest master of qgc.txt](https://github.com/mavlink/qgroundcontrol/files/822522/latest.master.of.qgc.txt)
[image]

DonLakeFlyer, 2017-03-07:
Can you get output from ParameterManagerLog?

lvale, 2017-03-07:
@DonLakeFlyer We've discussed this problem today on the dev call and hopefully the solution is this https://github.com/ArduPilot/ardupilot/commit/e4c9255cb908ffbac329d6e3f728c16a68b340f2

DonLakeFlyer, 2017-03-07:
The log doesn't match what the output should be from the newest builds which have #4723 in it. What OS are you using. Maybe there is a daily update problem.

lvale, 2017-03-07:
I'm pulling and compiling locally on OSX with QT5.8, and that had #4723.

[image]

lvale, 2017-03-07:
I'm compiling ArduPilot master with the changes from Peter Barker and Tridge for the UART/DMA issue that might be the problem here, and we'll let you know how it goes.

lvale, 2017-03-07:
@DonLakeFlyer @peterbarker 

So I compiled ArduPilot Copter with the latest commit from Peter and flashed the PixRacer (using the USB cable).

Also pulled the latest QGC from master and compiled with QT5.8

Disconnected the USB cable, rebooted the PixRacer and set up the WiFi connection to the PixRacer via  WiFi (Mac OSX 10.12.3).

QGC connected via WiFi and began the initial downloading of parameters - however sometimes it gets "locked" when just a few parameters (5/6) 

Restarted the process and the downloading of parameters restarted , BUT when the parameter download finishes QGC crashes.... see crash1.rtf

When restarting QGC I managed to save the tlog - crash_log.mavlink
[crash1.rtf.zip](https://github.com/mavlink/qgroundcontrol/files/825943/crash1.rtf.zip)
[crash_log.mavlink.zip](https://github.com/mavlink/qgroundcontrol/files/825944/crash_log.mavlink.zip)

lvale, 2017-03-07:
Retried again and managed to save a piece of the log while downloading the parameters, but still crashes when it finishes

[1.txt](https://github.com/mavlink/qgroundcontrol/files/825958/1.txt)

lvale, 2017-03-07:
Just adding an App Log from a USB connection. Same board, same firmware and same QGC
[usb_log.txt](https://github.com/mavlink/qgroundcontrol/files/825974/usb_log.txt)

peterbarker, 2017-03-07:
On Tue, 7 Mar 2017, Luis Vale Gonçalves wrote:

> Restarted the process and the downloading of parameters restarted , BUT when
> the parameter download finishes QGC crashes.... see crash1.rtf

Yes yes - but you haven't told us whether the parameter download was fast 
or not!  :-)

lvale, 2017-03-07:
@peterbarker it is fast :) until it crashes :)

DonLakeFlyer, 2017-03-07:
I have to see if it will repro for me on direct USB connection. My pixracer wifi is toast.

DonLakeFlyer, 2017-03-07:
> It is fast :) until it crashes :)

Story of my life...

lvale, 2017-03-07:
@DonLakeFlyer On USB everything is fine and dandy :)

NaterGator, 2017-03-08:
Are we blindly trusting data coming over the link to index into an array or list somewhere? Sounds like corrupted data over the WiFi link triggering a segfault from out of bounds access. 

Slow is smooth and smooth is fast!

DonLakeFlyer, 2017-03-08:
> Are we blindly trusting data coming over the link to index into an array or list somewhere?

Really?

NaterGator, 2017-03-08:
I reviewed it and couldn't find anywhere that corrupted data over the link would index randomly into a vector-style data structure. Worst case we'd get a miss in a hashtable and qt would create a default value for us.

DonLakeFlyer, 2017-03-08:
It's a wierd Qml thing (big surprise there) which I haven't figure out yet.

DonLakeFlyer, 2017-03-09:
Hmm, from looking at the last text log it may be declaring load complete incorrectly. Let me look at the debug output and see if I can add some better logging. 

@lvale What OS are you running on?

DonLakeFlyer, 2017-03-09:
@lvale Can you do another run with only ParameterManagerLog? Don't turn on the other more verbose parameter logs.

DonLakeFlyer, 2017-03-23:
This has been a real pain. I can't seem to get toolset set up to re-flash my own ESP 8266's for testing. So instead Gus is sending me on of his so I can make forward progress...

lvale, 2017-03-23:
Just tested after clearing the preferences from .config and fresh from master QGC

Parameter downloading completed without major issues although some retries occurred

Here's the app log

[log_ok.txt](https://github.com/mavlink/qgroundcontrol/files/866503/log_ok.txt)

closed 2017-03-23 by lvale

DonLakeFlyer, 2017-03-24:
Ah, good news finally. Your connection is really lossy though!

lvale, 2017-03-24:
@DonLakeFlyer Well the drone is at 30cm from the Macbook :)

[image]