mavlink/qgroundcontrol #4654: RTSP Video Stream Crash QGC

vsobalski, 2017-02-28:
Hi,
I’m experiencing some difficulties using the RTSP of my IP camera. The ground station works great, but when I change the RTSP Url from the general tab, connect my stream, and then restart QGC, the station crashes.
If I connect my stream (plug the camera back in) after QGC is connected, the station crashes shortly after reconnection. The stream works on VLC and has the following settings: 
URL: rtsp://10.0.0.150:554/live.sdp
Type: H.264, 1280*800@30FPS
Constant Bit Rate of 3Mbps
The Stack Trace is as followed (using Debug View):

```
1	0.000000	5420	QGroundControl.exe	Settings location "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true
2	0.016546	5420	QGroundControl.exe	Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
3	1.425577	5420	QGroundControl.exe	MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/aosl/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs"
4	2.335821	5420	QGroundControl.exe	Map Cache in: "C:/Users/aosl/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db"
5	3.744247	5420	QGroundControl.exe	Orphaned log file count 0
6	3.839707	5420	QGroundControl.exe	LinkManager mavlinkStatus:channel:flags 0x17cc15c 1 2
```

I am using the latest build released: V3.1.3
Thanks in advance for your help,
Cheers,

Vincent

NaterGator, 2017-02-28:
That should have been fixed by #4613, can you try a recent build from master?

vsobalski, 2017-03-01:
I just compiled the latest version, and it has been fixed. It is working really well now, thanks for your help.

closed 2017-03-01 by DonLakeFlyer