mavlink/qgroundcontrol #4655: Video recording stops when connection times out

NaterGator, 2017-02-28:
If the video link is operating at the fringe of range and the link times out while recording it will restart when back in range but the recording is stopped.

Ideally we'd have a separate clock source for the recording branch of the gstreamer graph and leave that portion of the pipeline playing while we repair the link to the network source. There would be gaps in playback representative of the dropped connection and recording will stay active.

jaxxzer, 2017-04-12:
Ok, the current behavior is by design. I think this is a good idea, and could be relevant to concerns with editing and post-processing other data. I can attempt to implement this behavior when I get a chance.

NaterGator, 2017-04-12:
@jaxxzer I have implemented this behavior in another project. I'm currently doing bring up on an product we're planning to launch imminently which is an Nvidia TX1/2 module + PX4 autopilot stack and should have the HEVC encode pipeline done by the end of the week. Once I have that knocked out I'll take a swipe at this and see if my code from the other project ports cleanly in here, as well as implement better gstbin support for dynamic pipelines.

jaxxzer, 2017-04-12:
@NaterGator That sounds awesome, thanks.

DonLakeFlyer, 2017-05-09:
Is this something that needs to go into 3.2?

jaxxzer, 2017-06-08:
I've heard that this can happen when QGC takes an exceptionally long time (5-6 seconds) to switch between tabs on the setup page. I've noticed that the outgoing MANUAL_CONTROL interval seems to bog down when switching between views as well.

DonLakeFlyer, 2018-06-18:
@jaxxzer Is there a fix needed here?

DonLakeFlyer, 2018-06-20:
@jaxxzer Reopen if still a problem here.

closed 2018-06-20 by DonLakeFlyer