mavlink/qgroundcontrol #4664: Console logs truncate numbers

jakehewitt, 2017-03-01:
It seems that the logging is truncating numbers to be at most 6 digits, for example this cherry picked output:

```
SurveyMissionItemLog: Coordinate  QGeoCoordinate(34.2684, -119.146)
. . .
SurveyMissionItemLog: coveredArea  34815.4
```

In both instances the numbers being logged are the results of truncating the numbers. Is this behavior inherited from the qt logging system or was this behavior configured? I guess more importantly, is there anything I can do to enable logging the full number?

NaterGator, 2017-03-01:
Unfortunately it is a shortcoming of the logging system. We'll need to handle this on case-by-base basis

DonLakeFlyer, 2017-03-01:
@NaterGator Could you put a bug in on Qt for QGeoCoordinate qDebug output? Should be 7 digits past decimal.

NaterGator, 2017-03-01:
Done in #4666.

closed 2026-10-05 by github-actions[bot]