mavlink/qgroundcontrol #4666: QGeoCoordinates used in debug output need to output 7 digits of precision

NaterGator, 2017-03-01:
Would be nice for this to be handled app-wide.

DonLakeFlyer, 2017-03-01:
This needs to be a Qt bug. We don't have control over qDebug() << QGeoCoordiante(...) output as far as I know.

NaterGator, 2017-03-01:
I actually think we can do this on our own. Working on it now...

DonLakeFlyer, 2017-03-01:
Awesome, thanks Nate

NaterGator, 2017-03-02:
I've played around with this on and off, and while I have been able to come up with app-wide fixes they are decidedly inelegant and abuse platform specific behaviors. It simply isn't worth the headache, so upstream this goes: https://bugreports.qt.io/browse/QTBUG-59259

DonLakeFlyer, 2017-03-02:
Thanks for trying Nate

jakehewitt, 2017-03-02:
Thank you Nate. Is there any temporary hacky way I could log the full numbers?

dogmaphobic, 2017-03-02:
If I'm reading this correctly, you could have a function that takes a `QGeoCoordinate` and spits out a `QString`. You can format it any way you want it.

```
QString QGCApplication::qGeoCoordinateToString(QGeoCoordinate& geo)
{
    QString txt;
    txt.sprintf("%f %f %f", geo.latitude(), geo.longitude(), geo.altitude());
    return txt;
}
```

In the code you would use it instead of feeding it directly to `qDebug`:

```
qDebug() << qgcApp()->qGeoCoordinateToString(_homePosition);
```

NaterGator, 2017-03-02:
There's a few approaches to attacking this... you could also just shift the [qSetRealNumberPrecision](http://doc.qt.io/qt-5/qtextstream.html#qSetRealNumberPrecision) manipulator right after calling qDebug() and not need to dereference the qgcApp pointer. 

I played around with replacing the stream operator at the linker, manipulating the debug macros, etc. There's not one silver bullet to solve this across all platforms transparently.

DonLakeFlyer, 2017-03-02:
Many different way to hack in one-off changes. No way to do it without having to change all references in code.

closed 2024-06-02 by HTRamsey