mavlink/qgroundcontrol #4667: "communication lost" trigger

mhugentobler, 2017-03-01:
When sending HIGH_LATENCY messages via a high latency link from a UAV to QGC it takes usually more than 3.5s before that message arrives, triggering the "communication lost" warning. As it is implemented now, you do not regain communication once the next HIGH_LATENCY message arrives.

DonLakeFlyer, 2017-03-01:
@mhugentobler  Can you verify this?

dagar, 2017-03-01:
@DonLakeFlyer this is something I've been meaning to discuss, so might as well bring it up here. The point of HIGH_LATENCY is to have a single 50 byte mavlink message you can send over something like Iridium SBD (short burst data) and only get a few messages through every minute.

So the outstanding issues I'm aware of are
 1. basic support for HIGH_LATENCY in QGC (relatively easy)
 2. keeping the connection "active" with HIGH_LATENCY only and no HEARTBEAT
 3. can we create some kind of basic vehicle without first receiving a HEARTBEAT? Often the situation will be first connect over local radio, then expect Iridium to takeover once the main radio is lost, but being able to use only HIGH_LATENCY in some limited capacity could be valuable.
 4. QGC probably needs a special connection type or awareness to treat the connection differently. Last time I used this it was ~ 10 cents per 50 Byte message.
    - no HEARTBEAT
    - no param or mission sync
    - no MANUAL_CONTROL
    - bonus would be features that prevent you from leaving full AUTO modes

If there's interest in tackling these issues now I can help. @DonLakeFlyer figuring if we can handle 3) would be helpful.

DonLakeFlyer, 2017-03-01:
I can fix the comm lost stuff. But handling the concept of high latency connection for all communication and turning things off/on is a huge piece of work. That's all over the place. I don't see that something for this release.

What identifies a high latency message so I can trigger off it for comm lost?

dagar, 2017-03-01:
We're talking about a single mavlink message called HIGH_LATENCY. http://mavlink.org/messages/common#HIGH_LATENCY

On the PX4 side enabling Iridium on telem2 starts mavlink at a very low rate streaming only that message and nothing else.

I wasn't suggesting it for this release. With your input to unblock a few parts I was going to gradually expand support. The idea of turning things off/on is only needed for 2 way communication.

DonLakeFlyer, 2017-03-01:
Ah, I've never seen that before (HIGH_LATENCY). Now I get it. Should be easy enough to plumb that through. Creating a vehicle from it as well is trivial, but the problem you will run into immediately is that QGC is going to go into it's Vehicle connect sequence asking for params, missions and so forht and then it's going to get quite annoyed.

dagar, 2017-03-01:
I wasn't sure how to create a vehicle without the HEARTBEAT (type, autopilot, etc). 

For now we could at least note all of these issues and any ideas for how to handle them eventually.
https://github.com/mavlink/qgroundcontrol/issues/4121

DonLakeFlyer, 2017-03-01:
> I wasn't sure how to create a vehicle without the HEARTBEAT (type, autopilot, etc).

Oops, I looked to fast. If that's missing it won't be possible. Unless you create some settings somewhere that says create this type of vehicle if you see one.

dagar, 2017-03-01:
@DonLakeFlyer I think that would be fine for how I intend to use it. The alternative is an updated mavlink message that we try to squeeze the bare minimum vehicle creation info into. Let me know if you feel strongly either way.

For QGC in the case where you only receive HIGH_LATENCY do you think some very basic crippled support is viable at this point (with some work)? No params, mission, knowledge of vehicle or autopilot type. Just enough info to display the vehicle location on the map, update the attitude indicator, etc. Eventually adding basic support to send single COMMANDs to reposition, land, RTL, etc.

DonLakeFlyer, 2017-03-01:
If possible I'd mod the HIGH_LATENCY message to include the two missing fields (type, autopilot). Then I can create vehicles from it. Does it include fligth mode like heartbeat. Not sure if that's really needed.

dagar, 2017-03-01:
It's tough because adding more fields pushes it over 50 bytes (billing
increment). Copying base_mode, custom_mode, and landed_state seem a bit
wasteful in retrospect.

If I were to take another pass (add a new message and deprecate the
existing I'd guess) any other suggestions?
http://mavlink.org/messages/common#HIGH_LATENCY
Notes added to the top of this issue -
https://github.com/mavlink/qgroundcontrol/issues/4121

mhugentobler, 2017-03-02:
about issue #4671: this is how I have it implemented right now, just for HIGH_LATENCY(#234) only. You regain communication once the message(#234) arrives. 

about the list of @dagar 
1) I added a button to switch between telemetry and iridium communication. Issue #4564 hinders complete functionality.
2), 3) no problem with a call to Vehicle::_connectionActive() when you receive a message (see issue #4671). The thing is you still have no way to reconnect to the vehicle in case QGC crashes, while your UAV is far away(i.e. reachable by satellite only), due to missing heartbeat information in message #234 (as you mentioned as well). If I checked correctly 3 Bytes from the HEARTBEAT message are required to create the vehicle object. Might be an idea to add these to message #234 if that's not in conflict with the 50 Bytes limit.
4) agreed. Params and waypoints should be saved on the groundstation PC for missions via satellite communication (again, consider case of QGC crashing). It would be nice to be able to send single commands (like return to home, change a single waypoint, etc.) from QGC to the plane.

Looks like we work on similar things @dagar (we also intend to use Iridium SBD)

dagar, 2017-03-02:
@mhugentobler I'll have to come back to this later in March to bring QGC + PX4 Iridium closer to usable.
Let's keep in sync in these QGC (https://github.com/mavlink/qgroundcontrol/issues/4121) and PX4 (https://github.com/PX4/Firmware/issues/6228).

In particular we'll need to put some thought into what can be dropped in HIGH_LATENCY to make it a full HEARTBEAT substitute.

DonLakeFlyer, 2017-03-06:
Closing since this has moved to #4121

closed 2017-03-06 by DonLakeFlyer