mavlink/qgroundcontrol #4668: Laptop Display Issue

ldsanders2807, 2017-03-01:
QGC 3.1.3: It appears that for some reason, the mouse does not function on the center part of the airframe selection screen.  I can't select ANY airframe.  I can't select any airframe's selection list.  Or use the mouse roller to scroll down the screen. The mouse does work on the outside panels. And everywhere else on QGC.  The laptop is an HP Elitebook 8540p.  It works fine on my desktop.

DonLakeFlyer, 2017-04-13:
Does this still happen to you using a daily build?

ldsanders2807, 2017-04-13:
I have not tried any version other than v3.1.3.   What is the link for the daily build?

DonLakeFlyer, 2017-04-13:
Daily build links at bottom of page: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/en/download_and_install.html

ldsanders2807, 2017-04-15:
The frame selection drop down menus work and the mouse roller will scroll the screen with the daily build version.

 

Thanks for checking back on this!!

 

LaVerne Sanders

DonLakeFlyer, 2017-04-16:
Perfect. Thanks for reporting back.

closed 2017-04-16 by DonLakeFlyer