mavlink/qgroundcontrol #4686: Intermittent dropped connection with Xbee with BT bridge

Patnetpatnet, 2017-03-02:
When connecting to QGC Android and QGC Windows to Pixhawk (PX4) with Bluetooth as a bridge to xBee  I get a dropped connection when sending data, such as a switch position change. 

Setup is flight controller to xbee via TELEM1 port. Then xBee to xBee. Then Xbee to BT. Then BT to Android device or Windows PC.  15,200 baud rate throughout.

Wired connections are OK. 

Also unable to replicate issue with same radio setup on Tower/Android or or Mission Planner/Windows--no issues.

Siytek, 2017-03-06:
@Patnetpatnet sorry I missed your comment in the other thread about it working with Tower. Have you tried connecting the BT module directly to the TELEM port on your Pixhawk?

I have both of these telemetry links working fine:

My main telemetry for flying @ 57600bps:
QGC BT Android/OS X ~~> TBS Crossfire Tx ~~> TBS Crossfire Rx --> PX4 BST/I2C

My short-range wireless configuration link/PX4 console @ 57600bps:
QGC BT Android/OS X ~~> HC-06 --> PX4 TELEM Port

The difference between our setups is Xbee, which would make me think that your problem lies there.

Patnetpatnet, 2017-03-07:
@Siytek, thanks. Not tested BT/TELEM direct. Does not function wirelessly with QGC Windows and Android. Does function via USB. Please let me know your result when you test with Android.

Siytek, 2017-03-08:
I would test BT to TELEM directly, see if you get the same issue, let us know the result. BT HC-06 to TELEM using QGC on Mac works 100% for me. 

Android also works since the BT issue was resolved, I've just not tried flying it yet since getting my Android tab. I will post the results, probably about a month off being air worthy. My BT/xFire telemetry link gives me the correct position on the map and the horizon instrument and compass work fine, no problem with the connection, that's about as much testing as I have done.

I don't know the details about how APM and PX4 use Mavlink but there are differences. Tower/MP may be less fussy about dropped packets, although I'm guessing there and have never really used them. Could you be loosing data and Tower/MP just doesn't complain about it?

DonLakeFlyer, 2017-03-08:
FYI: QGC was pretty fussy about determining lost connections. Used to only use the HEARTBEAT message to determine if a connection was alive. Most recent builds (a week ago) now use any messages across the link to determine a link is still alive.

Patnetpatnet, 2017-06-02:
Changed telem baud rate throughout BT and Xbee now to 57600. Still getting the same issue.

DonLakeFlyer, 2017-06-04:
Are you saying that you are getting intermittent communication lost messages. Or is it that the communication link just goes down and stays down.

Patnetpatnet, 2017-07-24:
The daily build in Play Store works fine. The released version will not connect via BT and Xbee. Crashes the QGroundcontrol Android app.

Patnetpatnet, 2018-03-05:
I'm running into a similar problem again. QGC Android will not fully load all Parameters via Xbee. Works fine via USB. Like it's timing out as the green line gets to about 80%. Almost like it buffers some data and then gives up. Any ideas?

closed 2026-10-05 by github-actions[bot]