mavlink/qgroundcontrol #4688: Ubuntu 16.04 segfault at launch

jaxxzer, 2017-03-02:
The application segfaults at launch on Ubuntu 16.04, since switch to 5.8.

jaxxzer, 2017-03-02:
Tested with precompiled daily build, and compiling natively from source using QT 5.8. Compiling the same tree using QT 5.5.1 does not segfault.

jaxxzer, 2017-03-06:
Same problems on QT 5.7...

DonLakeFlyer, 2017-03-07:
I'm assuming you built yourself? Daily build drops are still on 5.8.

jaxxzer, 2017-03-07:
Yes, regarding 5.7 I built myself.

dagar, 2017-03-09:
@jaxxzer is this the appimage, tarball, or running straight out of Qtcreator?

jaxxzer, 2017-03-09:
AppImage, and QTCreator.

hbrezak, 2017-03-15:
I'm running Ubuntu 14.04 and tried DailyBuild from official site, both Linux AppImage and Compressed archive. Both cases result in segmentation fault. Leading up to seg fault:

`Settings location "/home/hrvoje/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/hrvoje/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
Map Cache in: "/home/hrvoje/.cache/QGCMapCache300" / "qgcMapCache.db"
Segmentation fault (core dumped)`

NaterGator, 2017-03-15:
@DonLakeFlyer what about those QtLocation private headers we have mirrored in the QtLocationPlugin? I don't think they've been updated to match our current version of Qt and should be scrutinized as a possible source of problems here.

dogmaphobic, 2017-03-15:
Those are only used for Qt 5.5.1. For all other versions, the normal (distribution) path is used.

DonLakeFlyer, 2017-03-19:
I built on 14.04 with Qt 5.7 and everything works ok. Need to switch build system to 5.7

DonLakeFlyer, 2017-03-19:
Fixed and tested daily build AppImage on both 14 and 16 ubuntu. No more crash. Although gstreamer doesn't seem to work on either. I'll enter a seperate issue for that.

jaxxzer, 2017-03-21:
Still getting a segfault on daily build 16.04, gstreamer has been uninstalled.

```
$ ./QGroundControl.AppImage
Settings location "/home/jack/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/jack/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
Map Cache in: "/home/jack/.cache/QGCMapCache300" / "qgcMapCache.db"
Orphaned log file count 0
LinkManager mavlinkStatus:channel:flags 0xfc4b48 1 2
Segmentation fault (core dumped)
```

DonLakeFlyer, 2017-03-23:
@jaxxzer I can't seem to repro this. If you can repro on something you build yourself can you get it in a debugger and take a look?

jaxxzer, 2017-03-24:
I get a segfault at launch 19/20 times (Ubuntu 16.04 only), sometimes it will launch ok.

Here is a stack trace, always the same, everything but the top level call to app->exec() is disassembly, so I'm not sure where to go from here in troubleshooting the issue.

[image]

Here is a log with --logging:full set.

[qgclog.txt](https://github.com/mavlink/qgroundcontrol/files/869286/qgclog.txt)

DonLakeFlyer, 2017-03-24:
That looks like and OpenGL driver bug.

[screenshot 2017-03-24 15 42 25]<https://cloud.githubusercontent.com/assets/8315108/24314783/ca11775e-10b9-11e7-806f-8ddd618d6877.png>

jaxxzer, 2017-03-24:
I have a third-party (AMD Radeon) graphics driver on this machine, do you think that could be the part of the issue? 

Interestingly, the application always starts (slowly) with valgrind memory analyzer. So looks like a race is going on.

DonLakeFlyer, 2017-03-25:
Windows has similar problems with bad OpenGL drivers. 

@NaterGator What's the thing to tell QGC to use software rendering instead? Is that available in Linux as well?

jaxxzer, 2017-03-27:
Working for me now on master, haven't touched the QT installation...

DonLakeFlyer, 2017-03-27:
I'm just guessing flaky opengl driver + QGC ui moving around making different Qml calls. Closing for now to see if it comes back,

closed 2017-03-27 by DonLakeFlyer

jaxxzer, 2017-09-12:
This is a problem with my graphics driver. @NaterGator is software rendering available on Linux?

DonLakeFlyer, 2017-09-12:
If you look in main.cc you'll see ifdef Q_OS_WIN code for -angle and -swrast? You should build your own and see if it works for you.