mavlink/qgroundcontrol #4691: Change compass heading display to 3 digits

marinesimulation, 2017-03-03:
The heading indicator numbers from 1-99 degrees are in a two digit format, can we chance these to three digits where we have a zero in front? ("045") This would bring it inline with aircraft and vessel displays.
[image]

Antiheavy, 2017-03-03:
I will second this request!

DonLakeFlyer, 2017-03-03:
Is such a big compass display critical? We were thinking of moving to something more like this:
[image]
Which has the compass and hud combined.

jaxxzer, 2017-03-03:
I am in favor of reducing the overall size of the display.

DonLakeFlyer, 2017-03-03:
Not that specific image necessarily. Just the concept of compass around attitude.

KevinKW87, 2017-03-03:
Combining the compass and the attitude I think would work out well. If that's the case, can we still get a three digit compass reading in the Widget Value area? It makes navigation easier for me when I do quick glances at the widget values. I have a rough idea what my heading is just by glancing at the first digit (0-1-2-3).
[image]

Antiheavy, 2017-03-03:
I like the idea of combined compass and horizon as long as the pitch/roll and heading are clearly visible from a couple/few meters away from the screen.  We don't have to read the numbers from that far, but need to see the orientation by tilting and turning your aircraft during pre-flight checks.  For UAVs I like circular compasses compared tape style.

I also support the original request to display heading as a 3 digit number where shown.

DonLakeFlyer, 2017-03-03:
> If that's the case, can we still get a three digit compass reading in the Widget Value area?

Unfortunately that would be way more difficult since that is all generic code which at the point it's being displayed doesn't know it's a heading. It would require formatting meta data support in the internal FactSystem. Doable but non-trivial.

DonLakeFlyer, 2017-03-03:
Maybe if the compass goes around this:
[image]
We could put three digit heading in big text towards the bottom center part of the attitude indicator.

DonLakeFlyer, 2017-03-03:
Quick Qml hacking:
[image]

jaxxzer, 2017-03-03:
@DonLakeFlyer In the intermediary term (at least), can we make this new style an option?

DonLakeFlyer, 2017-03-03:
I tend to think options are evil. 5% of the user base ever finds anything beyond the default setting.

DonLakeFlyer, 2017-03-03:
An image of it working for real:
[image]

jaxxzer, 2017-03-03:
@rjehangir Has reservations about this. Almost all of Sub navigation is done by heading, and it may be unclear/less wieldy for operators. Even if the option is not exposed, something that we could default to with firmware?

DonLakeFlyer, 2017-03-03:
That's a good argument. 

You can't default with firmware. But you can default with your own custom version of QGC using your own core plugin. This allows you to change the default on settings as well as hide settings among a whole pile of other things which allow you to create a third party branded version of QGC targeted to a specific vehicle.

DonLakeFlyer, 2017-03-03:
With your own QGCCorePlugin you can even go so far as replace the instrument panel entirely with something custom created by you.

DonLakeFlyer, 2017-03-03:
Ok, one more. You can even change the vehicle icon throughout QGC from the core plugin to be a sub looking thingy instead of a plane looking thingy.

DonLakeFlyer, 2017-03-03:
Here is a full screen shot. It leaves a lot of room for lots of values to show in the page control below. Implementing as a second variant. So old one is still usable if you want.
[image]

marinesimulation, 2017-03-03:
Here's one vote against combining the attitude and compass displays.  When operating an ROV we really don't use the attitude display all that much.  In fact, the BlueROV2 is the only mini sized ROV that includes an attitude display.  It may help in some rare situations but our primary focus, especially when operating in poor visibility, is the compass heading.  Unlike aerial drones, ROVs frequently operate in water that may have almost zero visibility.  Having a large, easily read compass display is critical for this type of operation.  The current compass heading display is just about perfect.  It's a good size and it's easily read.  I think removing it and combining it with another, less used display would be a big mistake and would make QGC much less functional for ROV operators.

DonLakeFlyer, 2017-03-03:
I'm putting it on an option. But I'm defaulting to combined. We'll see how it goes. Easy to back off after it's through.

DonLakeFlyer, 2017-03-03:
Fixed

closed 2017-03-03 by DonLakeFlyer