mavlink/qgroundcontrol #4706: Landing Pattern bugs, 2017-03-04 daily build

Antiheavy, 2017-03-04:
WIP Landing Pattern is coming along nicely.  Here are some bugs from today's daily build:

1) Clicking on the map after placing a Landing Pattern will place a new normal waypoint, even though the Waypoint tool is not shown active.  This also seems to break the recently added functionality where after placing a Landing Pattern the Waypoint tools are turned off.

2) The Landing heading behaves badly.  Typing in a number redraws the pattern with an incorrect heading and it seems inconsistent how it calculates, it's like a sign or direction is flipping or something.  The only headings that work correctly are 090 and 270.    Clicking and dragging seems to show the correct heading from Loiter-->Land.  After draging, if you place the cursor in the Landing heading field and press enter the pattern redraws at an incorrect position.

3) Heading is currently calculated from the center of the Loiter to the Land point.  It would be an improvement to calculate the heading based on the CW or CCW tangents.

For what it is worth we successfully flight tested this build today on our fixed wing with PX4.

DonLakeFlyer, 2017-03-05:
> Clicking on the map after placing a Landing Pattern will place a new normal waypoint, even though the Waypoint tool is not shown active. This also seems to break the recently added functionality where after placing a Landing Pattern the Waypoint tools are turned off.

Fixed

> The Landing heading behaves badly. Typing in a number redraws the pattern with an incorrect heading and it seems inconsistent how it calculates, it's like a sign or direction is flipping or [...]

Fixed

> Heading is currently calculated from the center of the Loiter to the Land point. It would be an improvement to calculate the heading based on the CW or CCW tangents.

No geometry on the weekends :)

AndKe, 2017-03-09:
Please, when  planning landing:
1:ask user to click on landing point.
2:ask user to click on direction to land from. 

entering heading is fine on airports, when heading is given to you, but for real terrain, with a decent landing area,  it's easier, and more reliable, to just calculate heading from what user points out in map.

Mockup below:  (heading and flightpath is set as user moves cursor after setting landing point)

[image]

Another method:   user gets to see  landing point , the approach, and gets to edit it by moving parts just like editing two waypoints

DonLakeFlyer, 2017-03-09:
I think that make sense. @Antiheavy ?

NaterGator, 2017-03-09:
@DonLakeFlyer FWIW this is how we do it too. We've also had the use-case of walking the vehicle itself or a beacon out to the desired landing point to capture the appropriate coordinate; just depends on the terrain.

DonLakeFlyer, 2017-03-09:
Wait a minute. Right now you click to set landing point. Then you can just drag the loiter to where you want? Maybe two clicks is just easier. Worried about all the "do this", "then do that" prompting required for multiple clicks.

AndKe, 2017-03-09:
Yes. Maybe two clicks is just messy, better to just let user click position, spawn the approach and then let user edit it list like other mission items.
Currently, all I get is a green label named land, what would be nice is:
[image]
Really visual, and easy to place in terrain.

NaterGator, 2017-03-09:
@DonLakeFlyer logically it seems to make sense to have click 1: loiter, click 2: landing position. In practice the pilot usually has to drag the loiter around a bit to ensure the tangent landing approach is aligned for the corridor (assuming terrain requires a corridor style landing). I don't think the distinction between 1 and 2 clicks is that great since my experience is that there is always some final drag-based fiddling required to line everything up. 

Having said that I understand what you mean about having obtuse interaction with the user through stuff like rigid click patterns; definitely we should make the case where the user ignores the second click intrinsically safe (like if focus is lost a default landing position is dropped). The alternative is just to leave it as-is and have a better UI hint that the loiter and landing points are both meant to be dragged.

NaterGator, 2017-03-09:
@AndKe you have rendering issues of some sort. Things should look similar to [this](https://github.com/mavlink/qgroundcontrol/pull/4703#issue-211823071).

DonLakeFlyer, 2017-03-09:
> Currently, all I get is a green label named land,

Thats a bug. You loiter and land point are at the same place. If you type in a distance in the editor you'll see both. Not sure exactly what causes yet for a repro. Since it comes and goes for me.

DonLakeFlyer, 2017-03-09:
I"m running out the door right now. Can someone put in the loiter/land in the same place Issue?

AndKe, 2017-03-09:
@NaterGator  nice ! - your screenshot is great

I found that the loiter is there, just had to zoom out :) 
The line between "land" and loiter is not visible until landing heading changes.

Bug: toggling "loiter clockwise" does actually change the heading.
the proper solution would be to move the center of the loiter (with one diameter, perpendicular to aproach), so that when changing loiter direction, would still result in the same approach headng.

Antiheavy, 2017-03-10:
Great discussion!  I think what @DonLakeFlyer has implemented is a really good first-go at this.  I am a fan of the currently implemented single click to place Land point then drag the loiter where you want it.  This approach works great for simple 2 waypoint landing patterns.  It gets more complex fast once we start talking about box pattern, J pattern, and other landing patterns...  one step at a time.

Some of @AndKe and @NaterGator suggestions are really good too.  I really like these suggestions:

> -given how easy it is to just move the loiter, it's no biggie...
I agree.

> just to leave it as-is and have a better UI hint that the loiter and landing points are both meant to be dragged.
Neat idea to add UI hints as long as it is not too cluttered.

> the proper solution would be to move the center of the loiter (with one diameter, perpendicular to aproach), so that when changing loiter direction, would still result in the same approach headng.
This would be great!

> We've also had the use-case of walking the vehicle itself or a beacon out to the desired landing point to capture the appropriate coordinate
This is our standard operating procedure too.

Also, I've received some good suggestions from other people:

- consider using FW_LND_ANG parameter from the aircraft to calculate a "Final" point (the start of the final decent).  I see @AndKe illustration above shows a "Final" point that is sort of this idea.  Maybe simple "pip" indicator on the line would be less cluttered and complicated than a actual waypoint, but still show where the final approach is estimated to begin.  If the user drags the Loiter too short based on the calculated position of that pip the final approach line could turn red as a indicator that the autopilot's mission feasibility checker is likely to reject it.

- consider calculating the initial distance between Land and Loiter based on FW_LND_ANG geometry, if connected to an aircraft.  If not connected to an aircraft, use a built-in default.

DonLakeFlyer, 2017-03-11:
I don't think doing fancy things based on FW_LAND_ANG is a good idea. It requires the ground station to know quite a bit about the mechanics of the firmwares fixed wing auto-land implementation (like this: https://pixhawk.org/users/fixedwing_autoland). Including stuff like flare and so forth or whether the vehicle has a distance sensor or not I would assume. It is much better to keep the ground station stupid to this sort of thing because it can change from firmware rev to firmware rev. Not to mention that fact that all of it is firmware specific.

AndKe, 2017-03-12:
I agree, do not base it too much on actual parameters.
All mine missions are planned before traveling to the site.  the AP is not connected during the planning.
However, descend rates are well known and documented in POH of the given RPAS.

If anything, I'd suggest to save, and default to last used landing distance, and loiter radius, altitude,  - for those with only one airframe type, that would save some work.

BUG:  When "Land"  is moved, gradient is not updated. (displaying the gradient between last WP and Loiter)  This gradient, is very important when terrain calls for a steep descend.

dagar, 2017-03-12:
[image] [Review QGC landing bug](https://trello.com/c/ZrjRHbUg/16-review-qgc-landing-bug)

AndKe, 2017-03-12:
@dagar FYI: the link you posted is private.

dagar, 2017-03-12:
oh, thanks trello. Deleting

DonLakeFlyer, 2017-03-23:
> I'd suggest to save, and default to last used landing distance, and loiter radius, altitude, - for those with only one airframe type, that would save some work.

That is how it works now.

DonLakeFlyer, 2017-03-23:
> BUG: When "Land" is moved, gradient is not updated. (displaying the gradient between last WP and Loiter) This gradient, is very important when terrain calls for a steep descend.

This is a tricky one. Since this is a complex item as opposed to a simple waypoint gradient will be from the loiter point only. The mission item status at the bottom doesn't know about the land point. Let me think about this a bit to see what I can do.

I think the rest of the stuff listed above is fixed.

DonLakeFlyer, 2017-04-13:
@Antiheavy Should gradient value be added to landing pattern editor and a computed value?

Antiheavy, 2017-04-13:
> Should gradient value be added to landing pattern editor and a computed value?

Short answer:  I can see how some would find it useful.  I prefer point #4 below (at least for PX4).

Longer answer: a few points to consider:

1) In the new mission summary stuff at top of the Plan page, the gradient from the previous waypoint is displayed, but gradients internal to the landing pattern are not displayed.  e.g. even if you click on the Land waypoint the gradient shown is actually from the Loiter back to the previous waypoint.  I'm not sure if this will be a point of confusion or not.  For what it is worth: I've noticed the landing pattern waypoint summary stuff doesn't "round trip" either.

2) PX4 MissionFeasibiliyChecker will reject a landing slope that is greater than FW_LND_ANG parameter.  (Actually, I don't know how it deals with breaking out on the tangent of a Loiter waypoint, I wonder if it actually calculates the geometry correctly? - question for Firmware folks I guess)

3) Displaying the landing gradient would be somewhat useful in someone being able to see if their glide slope was no steeper than what they know their aircraft's FW_LND_ANG parameter was set too.  Of course, they'd have to remember this.

4) It would be more useful if QGC could check the FW_LND_ANG parameter from the autopilot and somehow indicate the landing pattern is likely to fail MissionFeasibilityChecker.  Maybe by making the line between Land and Loiter points red or something if it is dragged too short.

5) PX4 doesn't actually fly the landing gradient between the loiter and land waypoints.  It internally calculates where to start the decent based as FW_LND_ANG.  It maintains level altitude up until that point.  So in this sense displaying a calculated gradient between the Loiter and Land waypoints could be a little confusing if someone were to think that was going to be the approach angle.  Then again, if users understand that gradient is not necessarily the same as flight angle then okay.

It has been a long while since I've used APM and I can't remember how it handles this or what would be most useful.

DonLakeFlyer, 2017-04-13:
Trying to outsmart the feasibility checker on the QGC side doesn't sound like a good idea. That could change out from under the ground station release to release.

> I've noticed the landing pattern waypoint summary stuff doesn't "round trip" either.

What do you mean by that. If you put in a landing pattern and read it back you don't get a landing pattern back just individual commands?

Antiheavy, 2017-04-13:
> What do you mean by that

Prior to upload:
[image]

After upload and download:
[image]

Antiheavy, 2017-04-13:
> Should gradient value be added to landing pattern editor and a computed value?

Also, here is a more direct answer to your question:   Our use case does not require display of the landing gradient at this time.  However, we would not object if it was added at some point.

DonLakeFlyer, 2017-04-15:
Ok. I'm going to call this good for now. We'll collect feedback through the Stable 3.2 release and update with new features as needed.

closed 2017-04-15 by DonLakeFlyer

kikislater, 2017-08-27:
Hi,
Since 3.2.4, this nice work and feature has gone in stable and beta. 
Could we still have it ?

DonLakeFlyer, 2017-08-27:
I don't understand what the ask is for?

kikislater, 2017-08-28:
Just where is this now? I couldn't find it in stable or beta release. Before I was able to play with it but really never tested on aircraft. Now I have a little time to test it. I was not connected to my wing maybe it's related... 

_Sent from my Xiaomi Redmi Note 4 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

DonLakeFlyer, 2017-08-28:
I went looking for it as well and couldn't find it. Realized i screwed up the pull: https://github.com/mavlink/qgroundcontrol/pull/5178.

kikislater, 2017-08-28:
Don't worry I'm here about this nice feature 😀