mavlink/qgroundcontrol #4712: Android store listing

dagar, 2017-03-05:
Daily build
 - store listing
 - build configuration options for package name
 - monotonic versioning
 - deploy non release builds directly to production
 - updated icon
 - warning in description and encourage regular users to try stable
 - link from website

Stable
 - travis-ci deploy stable directly to production
 - privacy policy
    - Your app has an apk with version code 313000 that requests the following permission(s): android.permission.CAMERA, android.permission.RECORD_AUDIO. Apps using these permissions in an APK are required to have a privacy policy set.
 - update screenshots

Both should be installable on a single device and not conflict.

DonLakeFlyer, 2017-03-05:
> Both should be installable on a single device and not conflict.

That's not quite doable with any of the other builds. Settings are shared and so forth. Windows uninstalls previous build. Don't see why we need it for Android.

DonLakeFlyer, 2017-03-05:
Related to #4593

dagar, 2017-03-05:
That was something I was hoping for actually. Then I can have stable installed for actual use, while keeping up with development. With different package names they'll be installable simultaneously and with the way android apps are sandboxed the settings directories should be fine, but I'm still going to double check.

DonLakeFlyer, 2017-03-25:
@dagar We really need to get this up and running. I keep having to deal with people running into problems which are already fixed. It's not possible to do side by side install since the Android manifest is the same for both builds. Also can you set up the daily build as an opt-in beta only such that it doesn't show up when you do normal google play searches? Let me know if you have time to do this, otherwise I"m going to have to figure it ou myself.

dagar, 2017-03-25:
I was dragging my feet on this looking for a clean way to change the app and package name based on being a release build or not. For now I can do something quickly in a script to update the AndroidManifest.

DonLakeFlyer, 2017-03-25:
That won't fix the full problem. You will still have settings collision.

dagar, 2017-03-25:
I'll double check, but I believe the settings are in the application data directory (per package).

DonLakeFlyer, 2017-03-25:
Oh yeah, you're right

DonLakeFlyer, 2017-03-26:
Hmm, I was just looking around Play store and Tower is listed with both Tower and Tower Beta. Maybe we shouldn't require an opt-in and just make daily a regular build?

dagar, 2017-04-09:
https://github.com/mavlink/qgroundcontrol/pull/4947

DonLakeFlyer, 2017-04-14:
Done

closed 2017-04-14 by DonLakeFlyer