mavlink/qgroundcontrol #4716: Parameter download takes a very long time

dogmaphobic, 2017-03-06:
There is something funky going on at the end. It goes though 95% quickly but it takes up to 5 minutes to reach 100% and complete.

bkueng, 2017-03-06:
https://github.com/mavlink/qgroundcontrol/pull/4717 improves this

dogmaphobic, 2017-03-06:
#4717 definitely improves it but I bumped into an edge case that causes the parameter download to simply stop and never resume.

Every now and then (less often than not), but only during the parameter download, I get a "Communication Lost" quickly followed by a "Communication Regained". When that happens, the parameter download stops and does not resume.

bkueng, 2017-03-06:
Did that happen before #4717 as well? On lost communication you will run into timeouts more quickly, but the retry logic should still handle it properly. Maybe we need to pause the retry during communication loss. I don't know what other actions QGC is taking on a communication loss event.

dogmaphobic, 2017-03-06:
No, this is something new. I've been having quite a few "Communication Lost" events (one in five or so connections). Only after #4717 did the parameter download stopped and never resumed.

dogmaphobic, 2017-03-06:
> I don't know what other actions QGC is taking on a communication loss event.

I haven't take the to look either. What's important to point out is that this Lost/Regained occurs only during the parameter download and the elapsed time between lost and regained is minimal. What makes QGC assume the connection is lost? Is it a heartbeat timeout? Given the there is a stream of parameters coming, the link obviously didn't get lost. Is the firmware not sending enough heartbeats during a parameter send burst?

DonLakeFlyer, 2017-03-06:
This way my screw up. I was working on an ArduPilot param load probem and a change to the timeout times went through on accident.

DonLakeFlyer, 2017-03-06:
> I don't know what other actions QGC is taking on a communication loss event.

QGC takes no functional actions anywhere on communiation lost. That is only a visual indicator which allows the user to close the connection if they choose. Everything runs as normal.

DonLakeFlyer, 2017-03-06:
I have this fixed now. Pull will go into mainline soon. 

What it is showing me is that either the WiFi connection is lossy, or the firmware is losing requests either on the way in, or not reponding on the way out. Either one seems like a problem.

closed 2017-03-11 by DonLakeFlyer