mavlink/qgroundcontrol #4737: FW Landing complex mission item loiter/land overlap

NaterGator, 2017-03-09:
Spuriously the loiter and land positions will be dropped on exactly the same coordinate.  See #4706

DonLakeFlyer, 2017-03-11:
Anyone know if this is touch only?

Antiheavy, 2017-03-15:
I found a use case/bug in Windows build (KB/mouse, not touch) to cause this to happen repeatably:

1. Place a Landing pattern
2. Drag components around until "--.--" shows up in the Landing distance and Landing heading fields (bug).
               It seems random where this may actually occur so I end up just dragging Land point around randomly until I find a place that --.-- shows up.  I haven't been able to make --.-- show up by dragging the Loiter.
3. Hit [Enter] key.  "Invalid Value" pops up on the right side menu.
4. Press [Reset to default] button
Walla, the Loiter point jumps on top of the Land point.

DonLakeFlyer, 2017-03-15:
I think I'm going to change the landing pattern item to automatically add the pattern when you add the item. Then you just drag to repostion. This removes the strange user model and will likely make this problem go away as well.

DonLakeFlyer, 2017-03-15:
Also with this https://github.com/mavlink/qgroundcontrol/pull/4770 the --.-- should be gone now

DonLakeFlyer, 2017-04-14:
@Antiheavy I haven't seen this problem lately. I think it's gone now?

Antiheavy, 2017-04-14:
I've been using daily builds for the past month.  I have not seen this issue in a long time.   probably gone now?

DonLakeFlyer, 2017-04-14:
That's what I thought.

closed 2017-04-14 by DonLakeFlyer