mavlink/qgroundcontrol #4738: Segmentation fault (core dumped) when pressing on Radio tab

ChristophTobler, 2017-03-10:
I'm running latest QGC master and using a Snapdragon Flight Board (latest master). I start PX4 on the Snapdragon and the parameters are loading. As soon as I click on Radio it crashes (every time)

Here's the extended log
[QGC_crash.txt](https://github.com/mavlink/qgroundcontrol/files/833299/QGC_crash.txt)

ChristophTobler, 2017-03-14:
This `Unknown: 123...` is the reason for the crash
[image]

If I map all `Unknown: 123...` to a channel, I can select the Radio tab

DonLakeFlyer, 2017-03-14:
Awesome thanks. I think that should be enough of a hint to find it.

closed 2017-03-19 by DonLakeFlyer