mavlink/qgroundcontrol #4745: Windows daily not working

zenetio, 2017-03-11:
The daily HEAD:6b10539a5 2017-03-09 11:28:25 is not working on my machine. It does not connect. I verified that this version has about 90k while the last one, 3.1.3, has about 140k. Not sure if something missing. I had to revert to 3.1.3 where I dont see any issue.
I am using Windows 10 and when I connect via USB, it works fine but when I try to connect via Telemetry, nothing happens. The point is that Telemetry works fine with version 3.1.3.

DonLakeFlyer, 2017-03-11:
Boot QGC (with no telemetry radios connected):
* Settings/Console
* Click Set Logging
* Turn on LinkManagerLog and MultiVehicleManagerLog
* Connect your radio and start vehicle
* Wait a minute
* Back to Settings/Console Save App Log and attach here

DonLakeFlyer, 2017-04-14:
Closing due to no reponse

closed 2017-04-14 by DonLakeFlyer