mavlink/qgroundcontrol #4750: Error compiling Android QGC

melisab, 2017-03-11:
I kept getting no file found when compiling QGC Android. 

arm-linux-androideabi-g++: error: obj\moc_X4FirmwarePlugin.obj: No such file or directory 
makefile:1003: recipe for target 'debug\libQGroundControl.so' failed
mingw32-make: *** [debug\libQGroundControl.so] Error 1
18:34:59: The process "C:\Qt-android\Qt5.5.1\Tools\mingw492_32\bin\mingw32-make.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Android for armeabi-v7a (GCC 4.9, Qt 5.5.1))
When executing step "Make"

dogmaphobic, 2017-03-12:
I changed the title as this seems to be specific to that platform combination. I've never tried using Windows to build Android so I can't answer.

chinanewer, 2017-03-17:
我也遇到这个问题了，并且我的显示的是：
arm-linux-androideabi-g++: error: obj\moc_***: No such file or directory，但是这个文件并不是以obj结尾的，也就是文件名显示不完全

jilinggui, 2017-03-28:
year , i also encountered this error ,maybe it is debug of GCC 4.9 .when compile the debug ,you can change the filename.ex:change moc_PX4FirmwarePlugin.obj to moc_X4FirmwarePlugin.obj  in obj directory. but when compile release version , it does not work.

DonLakeFlyer, 2017-04-14:
Since QGC builds android fine on both Linux and OSX. This is some sort of configuration setup problem with Qt as opposed to a bug in QGC.

closed 2017-04-14 by DonLakeFlyer

AZOM, 2017-04-18:
I have the exact same issue under Win7...

> 14:01:34: Running steps for project qgroundcontrol...
14:01:34: Configuration unchanged, skipping qmake step.
14:01:34: Starting: "C:\Qt\Qt5.7.1\Tools\mingw530_32\bin\mingw32-make.exe" 
linking debug\libQGroundControl.so 
arm-linux-androideabi-g++: error: obj\moc_APMSensorComponent.obj: No such file or directory
Makefile:1110: recipe for target 'debug\libQGroundControl.so' failed
mingw32-make: *** [debug\libQGroundControl.so] Error 1
14:01:40: The process "C:\Qt\Qt5.7.1\Tools\mingw530_32\bin\mingw32-make.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Android for armeabi-v7a (GCC 4.9, Qt 5.7.1))
When executing step "Make"
14:01:40: Elapsed time: 00:05.

NaterGator, 2017-04-18:
I've seen this error before. Make sure you don't have spaces anywhere in the path to both the source and shadow build directories.

AZOM, 2017-04-18:
I don't have any spaces in those paths... 

> C:\workspace\qgroundcontrol
C:\workspace\build-qgroundcontrol-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug

DonLakeFlyer, 2017-04-18:
Here is what I would do:
* Try building an minimal android project using your setup. There should be some example you can use.
* If you can't get that to work. Then the place to ask questions is on the Qt forums.

jilinggui, 2017-04-20:
@AZOM yeah, I also encountered the same problem，but when i install it on my tablet with android 6，it work well. Have you solved this problem yet?

AZOM, 2017-04-25:
@jilinggui No I have not solved it yet. And as you said I am also using a Pixel C with Android 7.1.2. Still the issue is with building the whole Qt-project itself on a Windows environment.
Of course I would prefer using a linux/macOS environment, but I am forced here at work to use Windows :/

XYue, 2017-07-03:
I encountered this problem as well, and i used Qt5.7 under windows.

xgdwhy, 2018-11-11:
I encountered this problem as well, and i used Qt5.11.0 under windows 10 x64.and use master daily stable brach.i tried to rename moc_xxx.obj to match the missing file ,and it compile  success in ndk-10e in debug mode, fail in ndk-10e and ndk-16 in release mode.