mavlink/qgroundcontrol #4751: Planning tools keyboard commands request:

AndKe, 2017-03-12:
When an mission item is highlighted, (blue frame) , please make "Up Arrow" & "Down Arrow" select next/prev mission item

Please make "PgUp" / "PgDn"  move an item up/down the list.
Please assign
Ctrl+S  = save to file 
Ctrl+L = load from file 
Ctrl+W= write to AP (upload mission)
Ctrl+R= read from AP (read mission)

Same four shortcuts are very useful in full parameter list as well.  the idea is

DonLakeFlyer, 2017-03-12:
Will consider for 3.2

Antiheavy, 2018-02-04:
I'll piggyback on this my request for +/- keys to zoom map:  https://github.com/mavlink/qgroundcontrol/issues/6055

ricardohalfeld, 2023-02-07:
Are there any keyboard bindings for QGroundControl? It would make the experience a lot better.

Davidsastresas, 2023-02-07:
I don't think we have right now, but it would be a great addition. I will take a look at it in the incoming weeks.