mavlink/qgroundcontrol #4752: Reducing visual complexity

LorenzMeier, 2017-03-12:
Some of the menus have title bars which consume screen real estate (I'm running out of screen space on a 13" MacBook, so its not like we would have excessive space). Here is an example for a before and after version of a UI element with title bars removed.

One example of a title bar which does not add value for users (in particular since the functions of the tool buttons are "planning", "uploading" and "map adjustments").

[image]

Before:

[image]

After cleanup (not resized, since I did this just in the screenshot):

[image]

DonLakeFlyer, 2017-03-12:
The right side editors scroll up and down and they automatically position themselves to the selected waypoint when you click on them in the map. Not sure what this accomplishes other than removing helpful information?

DonLakeFlyer, 2017-03-12:
Some tweaks still can be done though without using the helpfulness. For example when you are actually connected to a vehicle there is no need to tell the user what Firmware and Vehicle they have. That is mostly for offline editing. I'll change.

DonLakeFlyer, 2017-03-12:
Here is what it looks like now:

Offline Editing mobile (I figure nobody is going to type on lat/lon on mobile):
[image]

Offline Editing desktop (same as before):
[image]

Vehicle connected editing mobile (no need to show firmware/vehicle):
[image]

Vehicle connected desktop (no need to show firmware/vehicle):
[image]

DonLakeFlyer, 2017-03-12:
There also is not enough "air" in these editors. Everything is right next to each other vertically which leads to looking more complex. We could chose to add more seperation and use more vertical real estate. Survey is a possible good example of this.

DonLakeFlyer, 2017-03-12:
I'll keep this open as a work in progress. The current pull optimizes for vertical real estate which I think in reality is a mistake.

DonLakeFlyer, 2017-04-14:
It's better now. Not necessarily great. But good.

closed 2017-04-14 by DonLakeFlyer