mavlink/qgroundcontrol #4753: After removing all items from a mission, view centers at min zoom level (it should center at max)

LorenzMeier, 2017-03-12:
As visible below. The other entering tools are also not zooming in.

I would propose the following: If zoom level is already close to max (which is a hint the user is seeing the vehicle sufficiently well), leave as is. If the zoom level is however not set or very high, zoom in so the user can see the local context.

If this happens to you in the field you're busy for the next 20 seconds zooming in with two fingers.

[image]

DonLakeFlyer, 2017-03-12:
There is still a Center tool zoom problem which happens but I can't seem to find the repro for right now.

DonLakeFlyer, 2017-04-13:
This is all different code now

closed 2017-04-13 by DonLakeFlyer