mavlink/qgroundcontrol #4758: Temperature is -1 , yellow popup display old waypoint

AndKe, 2017-03-12:
Feel free to close this issue, or request more data.
There are two possible issues here: 
1: temperature in the primary flight display area is displayed as "-1"
2: the yellow popup , gets stuck displaying one "reached waypoint..."  message, 

Please note that this is a arduplane .tlog playback, not truly live data. (so QGC does not have duplex communication.)

[image]

jaxxzer, 2017-04-18:
I can't see what the temperature says, but can you provide a method of reproducing the issue? What firmware version of APM and what QGC commit? Also does the temperature issue happen when actually connected to a live ap, or only when doing log playback?

AndKe, 2017-05-08:
Just checked with the log in the screenshot, and current 3.7.1 tlog , unable to reproduce anymore.

closed 2017-05-08 by AndKe