mavlink/qgroundcontrol #4781: Tagging photos fails without feedback

lvale, 2017-03-15:
Issue tagging photos where I can see on QGC console that the EXIF parsing fails.

@DonLakeFlyer has a demo bundle to test

DonLakeFlyer, 2017-03-15:
@birchera Can you look at this? Let me know and I'll get the data to you.

birchera, 2017-03-16:
@lvale can you tell me what error message you get? And can you post the console output for the topic `GeotaggingLog`?

Do you have multiple/all failing datasets, or is it just that one?

lvale, 2017-03-16:
Hi @birchera 

Issue happens with several data sets (common between them is camera - Sony A7 and PX4 version )

The Console message is `Could not decode creation time and date ("")"`  at ExifParser.cc:40

birchera, 2017-03-17:
Looks like there are multiple problems.

It cannot find the create date in the exif header of your images. Can you verify that it is there?

Also it cannot extract the triggering events from the px4 log. Do you use custom logging settings? Do you log gpos and the triggering messages?

lvale, 2017-03-17:
This is a dump of the exif data the Sony's have
[exif_dump.txt](https://github.com/mavlink/qgroundcontrol/files/850264/exif_dump.txt)

I'm not using the triggering events on the PX4.

birchera, 2017-03-17:
To clarify the workflow: You have to trigger the camera from the firmware. Then the locations at which this is done are logged in the log. From there the triggers are extracted and using the timestamps of the images from the camera, a matching algorithm tries to assign them to images.
If I understood you correctly, you have a different way of triggering the images, therefore the current functionality cannot assign georeference to the images.

lvale, 2017-03-17:
But even so it should not fail extracting the data from the EXIF ?

DonLakeFlyer, 2017-03-23:
@birchera Any update on reading the exif data?

birchera, 2017-03-23:
I can see that the required field is there in the dump. To see what is going wrong we would need to look into the binary data and how it is read out. Maybe there is some endian conversion missing somewhere. As far as I remember I only had the chance to test it on a range of big endian systems.

lvale, 2017-03-23:
I can easily extract the EXIF data using EXIFTOOL or other "regular" apps like OSX Preview, Aperture, LightRoom

DonLakeFlyer, 2017-04-15:
@birchera Any update on this. Wrapping things up for a stable 3.2 release.