mavlink/qgroundcontrol #4787: Mission item "Takeoff", dangerous bug.

AndKe, 2017-03-16:
Just discovered that a "takeoff" , sets absolute altitude of 50m 
That's well below ground level.

Additionally: takeoff item should have editable altitude.
When mapping an valley, or any place with a hill nearby, it's preferable to fly from the high elevation, just to ensure good telemetry link over a much bigger area.  In such cases, there is no point in climbing 50meters, (just to descend soon after.)

AndKe, 2017-03-17:
Extra info; it also affects a good takeoff command loaded from the plane - will be cahnged to ABS when uploading.  (even if the takeoff mission item was not touched)

DonLakeFlyer, 2017-03-19:
> takeoff item should have editable altitude.

@AndKe Which firmware? PX4 or ArduPilot?

AndKe, 2017-03-19:
ArduPilot.  - the result should be equally bad on both, unless PX4 treats anything as relative altitude.

DonLakeFlyer, 2017-03-19:
ArduPilot doesn't support takeoff altitude. Confused?

DonLakeFlyer, 2017-03-19:
Oh, wait I was looking at ArduPilot code wrong. Sorry.

AndKe, 2017-03-19:
Just check the fact that mavlink does have altitude type flag also for
takeoff, and - in QGC - creating Takeoff does not let you let you select..
Also:
I did also: (connect AP - Mavproxy - qgroundcontrol)
-Created takeoff in QGC
-uploaded to AP
-read back to mavproxy,
-in mavproxy: corrected to relative altitude.
-uploaded from mavproxy to AP
-read in QGC , (did not touch takeoff item)
-then just write mission from QGC to AP.
-takeoff was again modified to absolute altitude..

- yes, I do some checking before bitching :)

DonLakeFlyer, 2017-03-19:
Unfortunately not a simple fix because takeoff specifies an altitude but not a coordinate which screws up the concept of whether there is MAV_FRAME or now.

AndKe, 2017-03-19:
The yaw is also be important , when doing autotakeoff from runway ,
catapult or throwing is not so critical...

DonLakeFlyer, 2017-03-19:
Related to #4426

DonLakeFlyer, 2017-03-22:
@AndKe By tomorrow new daily builds show have this fix. Could you test?

AndKe, 2017-03-22:
yes, I am out flying, will do later today.

AndKe, 2017-03-22:
I was about to test, but when I compile current head , I get : 
```
 ./QGroundControl 
inotify_add_watch("/home/andre/.config/ibus/bus/88068ace76ef430096168750f1405ff6-unix-0") failed: "No space left on device"
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/.local/share/QGroundControl.org/QGroundControl/MAVLinkLogs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0) 
Aborted (core dumped)

```

Of course, there's plenty of space left,   - and yes, it's compiled clean, using 
```
QMake version 3.0
Using Qt version 5.7.1 in /home/andre/Qt/5.7/gcc_64/lib
```

DonLakeFlyer, 2017-03-23:
I found and fixed that problem in one of the pulls from yesterday. Tested on master and things worked fine.

AndKe, 2017-03-23:
Even scarier bug now, read an mission from ArduPlane , and you'll see all WP be at 0m altitude...

DonLakeFlyer, 2017-03-23:
Crap, sorry about that.

AndKe, 2017-03-23:
no harm done, it reminds me of the "need" for https://github.com/mavlink/qgroundcontrol/issues/4792

DonLakeFlyer, 2017-03-23:
Daily builds are usually really stable. Grid is going to be a while.

DonLakeFlyer, 2017-03-23:
I can't reproduce the 0 waypoint height problem. I tried with arduplane stable and master. Are you building QGC yourself or pulling a daily build. If pulling daily build which OS? May daily builds aren't updating correctly.

AndKe, 2017-03-24:
building myself, will reproduce later and provide steps (unless it was a
phenomenon extraordinaire on my end)  :)

AndKe, 2017-03-24:
Recompiled current master - works fine now. Thanks.

closed 2017-03-24 by AndKe

DonLakeFlyer, 2017-03-24:
Phew! Thanks for all the testing. Sorry for all the problems.

AndKe, 2017-03-24:
no problem, I am doing this gladly so one day I can use it professionally.