mavlink/qgroundcontrol #4790: build master version from source with error: "Telemetry save error"

CrazyPopLin, 2017-03-17:
I build the master version just right now with an error like "Telemetry save error" ;
The full Application Output:

> Starting /Users/Crazymumu/Develop/qgroundcontrol/DEV_QGroundControl/build-qgroundcontrol-Desktop_Qt_5_5_1_clang_64bit-Debug/debug/QGroundControl.app/Contents/MacOS/QGroundControl...
> QML debugging is enabled. Only use this in a safe environment.
> Settings location "/Users/Crazymumu/.config/org.qgroundcontrol/QGroundControl.ini" Is writable?: true
> Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
> MAVLinkLogManagerLog: MAVLink logs directory: "/Users/Crazymumu/Library/Application Support/QGroundControl.org/QGroundControl/MAVLinkLogs"
> Map Cache in: "/Users/Crazymumu/Library/Caches/QGCMapCache300" / "qgcMapCache.db"
> qrc:qml/MainWindowInner.qml:304:5: Type FlightDisplayView unavailable
> qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayView.qml:146:13: Type FlightDisplayViewMap unavailable
> qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewMap.qml:30:13: Cannot assign to non-existent property "acceptedGestures"
> Orphaned log file count 1
> Orphaned log file "/var/folders/6x/q9d6j429565cl0rt00qbq55h0000gn/T/FlightData333161.mavlink"
> QGCMessageBox (unit testing) "Telemetry save error" "Unable to save telemetry log. Telemetry save directory is not set."
> LinkManager mavlinkStatus:channel:flags 0x1039ea4a8 1 2

The App is fully white without any thing, and can't closed it. If you want to closed, it shows:

> qrc:qml/MainWindowHybrid.qml:25: TypeError: Cannot call method 'attemptWindowClose' of null

DonLakeFlyer, 2017-03-17:
You need to use Qt 5.7.1

CrazyPopLin, 2017-03-21:
sorry, I forget to review ReadMe.md. The new build use Qt5.7

closed 2017-03-21 by CrazyPopLin